publiclab/spectral-workbench #21: Problem with Spectralworkbench

bearacuda, 2015-01-07:
[image]
[image]
I have attached two screenshots from my Galaxy Note 3 smartphone.  I am using the smartphone DIY instructions and the back facing camera.  I found first that the online spectralworkbench utility does not work with both Chrome (back facing camera only displays 'black' with nothing covering camera!) and Opera.  I can only use it properly with current Firefox.  The screenshots come from the DIY spec/smartphone pointed at a perfectly white display on a 12" tablet.  Why when I perform the capture does NOTHING SHOW UP!  I have also tried to point it at a compact fluorescent, but same thing happens.  What am I doing wrong?
Thank you,
Jason

jywarren, 2015-02-10:
Hi, Jason - this is a bit mysterious; have you tried using your finger to manually select a different yellow cross section? Perhaps the yellow line is not actually positioned there, but doing it again manually would re-position it. Can you give that a try?

bearacuda, 2015-02-10:
Jeff,
     I did finally get it to work.  Not sure exactly what changed, but I
got results I could believe.  BTW, I used it try and determine a decrease
in blue light output from computer/cell phone displays using blue light
removing apps and screen covers.

Thank you,
Jason

jywarren, 2015-02-10:
Oh, very good to hear, and cool! I'm closing for now; we'll catch new bugs in the upcoming Spectral Workbench 2.0 revamp; check it out: http://publiclab.org/notes/warren/02-10-2015/planning-for-spectral-workbench-2-0

closed 2015-02-10 by jywarren