publiclab/spectral-workbench #23: Offline Use/RasPi Image

pedigoke, 2015-01-31:
This gizmo is small enough that it would be useful in field camps where there is no internet connectivity.  Making available a RaspberryPi image that permitted "off-line" usage with possibly a locally stored database of popular exaple spectrograms would greatly improve the usefuleness of this appliance

btbonval, 2015-01-31:
A simple way to do this:

1. Create an image that has internet access and a local web browser (headless FF might work, or run a full on X server with FF on the rPi).
2. Visit spectral workbench using the local web browser. Offline version is downloaded at that time.
3. Take the rPi offline and visit spectralworkbench.org to verify the offline version has been downloaded.
4. Burn a new rPi image from the above.

You'd only be able to use the rPi's local web browser, because it would use all the web browser caching stuff locally. Using an X Tunnel could allow access to the rPi's local web browser from outside of it, although that means a user needs to install SSH and an X client.

btbonval, 2015-01-31:
Local database full of spectra...

Yeah that's a different problem. We do need a good way to export our data.

btbonval, 2015-01-31:
Created #24 to track database spectral exports. It's a discussion topic that has come up in the past but apparently did not make it into a Github issue.

pedigoke, 2015-02-02:
I see that th NIST samples have JSON and XML export buttons.   It's a
thought....

pedigoke, 2015-02-02:
The browsers cache wouldn't capture all the ruby used for server side
processing.  I see that spectralworkbench is downloadable from GITHUB,
 Unless you have to authenticate against something on the 'net, that
version might do the trick I'll give it a whirl

jywarren, 2015-02-10:
Yes, you can run it locally; please tell us how installation goes, and if you have a RPi image to share, that'd be great too! Shall we close this for now, or wait until there's a pi image?

pedigoke, 2015-03-22:
I don't have a working Pi image yet... still a little held up trying to
recreate the hardware from what I can get locally down here in southern
Chile.  I may just order a kit from you to get the ball rolling.

I think as a general thing, a Pi image on the website would be outstanding,
and if I get a good one going, I'll post it.

jywarren, 2015-03-23:
I agree, it would be super. Thanks so much, I look forward to it!!