publiclab/spectral-workbench #24: Export spectra from spectralworkbench.org for external use

btbonval, 2015-01-31:
Because open data and portability are good things(TM)

We'll want to export some small amount of metadata regarding ownership and licensing for each spectra, and then the spectra itself (either as an image or as e.g. a CSV of processed values).

btbonval, 2015-01-31:
uhm, not sure if our analysis are compatible with mass spec, but this is an interesting reference of open data formats for use with mass spectrometers.
https://en.wikipedia.org/wiki/Mass_spectrometry_data_format#Open_formats

jywarren, 2015-02-10:
Hi, Bryan, just to be clear, how is this issue distinguished from the export formats available in the left side of each page: http://spectralworkbench.org/analyze/spectrum/43491

btbonval, 2015-02-11:
It was a request by someone to the web@ list. I'll have to look at that
later. It isn't showing up on the mobile version, which makes sense.

btbonval, 2015-02-11:
Ah I see the export functions now. I'm not sure what the person wanted who emailed web@, but I think it might be worth keeping this ticket open to investigate open standard formats for data of this kind.

The current export formats are custom and arbitrary based on how we store the data, right?

If the open standards aren't important, we can close this ticket.

jywarren, 2015-02-11:
I think the only standards are for csvs, and we do have an open ticket for
that.

btbonval, 2015-02-11:
Did you see the wikipedia link, posted in an earlier comment on this ticket, with a list of open standards of mass spec data?

JCAMP-DX is an ASCII based standard, ANSI-MS is based on netCDF (whatever that is), etc.

jywarren, 2015-02-11:
oh cool, i do remember that; was looking in email view only.

btbonval, 2015-02-11:
I can rename the ticket to be more appropriate for exporting standard mass spec formats. Under the current title of exporting spectra, this ticket is already resolved.

jywarren, 2015-02-11:
Cool, and let's disambiguate from the CSV requests in the other issue.
Thanks!

grindhold, 2017-04-08:
there is also the NUTS file format.
and i support this issue.