publiclab/spectral-workbench #43: store image captured and additional details in localstorage in offline mode

jywarren, 2015-02-10:
offer to upload when reconnected. @Sreyanth - I remember you had some part of this working, right?

Sreyanth, 2015-02-11:
I implemented the storage (saving and deleting) in localstorage here: https://github.com/Sreyanth/OfflineSWB/blob/master/js/spectrumstorage.js

Haven't implemented the upload when reconnect part as it needs a new controller to be written on the backend to receive this data, check for validity and authenticate the user, and then store the data into our database appropriately.

jywarren, 2015-02-11:
Do you think we could re-integrate your js code into this codebase? we
could then rely on whether you're logged in in this system, then access the
localstorage stuff via your code, and upload under the current username.

Sreyanth, 2015-02-11:
Yes. That can be done. Also, we should verify the validity of the data that is being posted to the URL controller because there might be some attackers who could possibly mess up with the data and create spam.

jywarren, 2015-02-11:
Most post requests require a login and CSRF token (
http://guides.rubyonrails.org/security.html#cross-site-request-forgery-csrf);
are you worried about real users, or anonymous ones?

Sreyanth, 2015-02-11:
Yes. CSRF is a must for POST requests in RoR and Django. So I have my plans.

First sending a GET request to get the CSRF and then immediately send a
POST. This way even expired CSRF token issue will also be taken care of.
What do you think?

I am assuming that if you want to upload, you need to be logged in. So an
account is compulsory. But given an OPEN ID integration, spamming won't be
that difficult.

jywarren, 2015-02-11:
If a version of this script is loaded by default when you access
spectralworkbench.org/capture (in offline or online mode) then the issue is
that you're probably looking at a cached version because of our offline
functionality via /index.manifest. So yes, good idea; you'll have to
retrieve a CSRF token via ajax. This is probably also when you can check if
you're online -- look for a 200 "success" response code. If you don't get
200, don't offer the user the ability to upload your saved spectra.

Sreyanth, 2015-02-11:
Yup. Sounds good. But I think there should be a better way of checking if
the user is connected to the internet than to keep pinging the main server
to get a 200. That would simply burden the browser activity.

A more robust solution is to use
var online = navigator.onLine;

See the related API section mentioned here
http://www.w3.org/TR/offline-webapps/
If a version of this script is loaded by default when you access
spectralworkbench.org/capture (in offline or online mode) then the issue is
that you're probably looking at a cached version because of our offline
functionality via /index.manifest. So yes, good idea; you'll have to
retrieve a CSRF token via ajax. This is probably also when you can check if
you're online -- look for a 200 "success" response code. If you don't get
200, don't offer the user the ability to upload your saved spectra.

> Yes. CSRF is a must for POST requests in RoR and Django. So I have my
> plans.
>
> First sending a GET request to get the CSRF and then immediately send a
> POST. This way even expired CSRF token issue will also be taken care of.
> What do you think?
>
> I am assuming that if you want to upload, you need to be logged in. So an
> account is compulsory. But given an OPEN ID integration, spamming won't be
> that difficult.
>
>
> > Most post requests require a login and CSRF token (
http://guides.rubyonrails.org/security.html#cross-site-request-forgery-csrf
> > );
> > are you worried about real users, or anonymous ones?
> >
> >
> > > Yes. That can be done. Also, we should verify the validity of the data
> > > that is being posted to the URL controller because there might be some
> > > attackers who could possibly mess up with the data and create spam.
> > >
> > > <
https://github.com/publiclab/spectral-workbench/issues/43#issuecomment-73907136
> > >
> > >
> >
> > <
https://github.com/publiclab/spectral-workbench/issues/43#issuecomment-73922292
> >
>
> >
>
> <
https://github.com/publiclab/spectral-workbench/issues/43#issuecomment-73923353

jywarren, 2015-02-11:
Oh cool, yeah. Still, to check if the user is logged in and to get a token,
we may have to make a request anyways.

Sreyanth, 2015-02-11:
Yep! We only need to send one request to get the token, eliminating
unnecessary burden on the server.