publiclab/spectral-workbench #73: Feature Request: Add location metadata to spectra image

kumavis, 2015-02-11:
I'm in Peru with Raul Yusta of UAV who is trying to deploy SpectralWorkbench in the amazon to collect pollution levels due to oil spills. He is having some difficulty training the local monitors to manually manage the data / add GPS information. He was curious if it would be possible to capture location data and embed that in the spectra image file.

Let me know if I can provide more information.

jywarren, 2015-02-11:
Hi, are you uploading images or live capturing? There is an issue open for uploading here: https://github.com/publiclab/spectral-workbench/issues/53 and we could expand that to include debugging the live capture interface too.

jywarren, 2015-02-11:
I'd like to merge this into #53 for now; @kumavis, if you go over there, please speak up if you need geolocation debugged on capture as well. Thanks!

closed 2015-02-11 by jywarren

kumavis, 2015-02-12:
We're live capturing. it sounds like you are already embedding GPS information into the images... is that right? if so -- awesome! I am not yet that familiar with your system and just wanted to communicate the desires of a less technical consumer of your work.

kumavis, 2015-02-12:
Here is a little more information about what we're doing https://github.com/publiclab/spectral-workbench/issues/77