publiclab/spectral-workbench #82: Controlling the size of related sets listed on the left hand side

Sreyanth, 2015-02-18:
The sets listed on the left hand side (like in spectralworkbench.org/search/led) are causing the page to load slower and compelling the page to have a lot of white space in the bottom. The related sets could be confined only to some 5-8 with a button to explore more sets.

jywarren, 2015-02-18:
we could also show them in 2 tabs in the same column, like on the dashboard

Sreyanth, 2015-02-18:
Yes, but still we will have to load all the data on to the browser right? I used the developer tools in Chrome to see how much content spectralworkbench.org/search/led is loading, and it turns out to be more than 14.6 MB 

I don't see the point of loading so much content when we still browse away to `/sets/show/id` anyways. I think we should some how optimize the page.

jywarren, 2015-02-18:
Oh yes, certainly we want to paginate the sets to save data transfer etc.
By putting it in a tab, we can also ajaxify the loading, so it doesn't load
at all unless you press that tab.

Sreyanth, 2015-02-18:
Right. We could also use the loaded data to be used to display the set instead of pointing to `/sets/show/id` right?

jywarren, 2015-02-18:
Yeah, maybe we start with optimizing the page itself, but an alternate
issue could be optimizing set list display by not having to load raw data
for each of the spectra in list display? There are other ways to represent
a set than to have the mini-graph.

Sreyanth, 2015-02-18:
Yeah. Exactly what I wanted to say! :smile:

jywarren, 2015-03-02:
Somewhat fixed using pagination in acf2a07a713df9cb57e3df418eb1b99835932a0c

still want to:

* [ ] optimizing set list display by not having to load raw data for each of the spectra in list display
* [ ] possibly ajaxify sets loading, move sets display into a tab

jywarren, 2015-03-02:
Oh, sorry, my fix was for profile page only. Now optimizing the search page; this may help w/ downtimes and hanging. that was an awful query.

jywarren, 2015-10-09:
Closing this as it's been revised in the 2.0 interface for spectrum display; we could revisit for front-page sets display if we like.

closed 2015-10-09 by jywarren