publiclab/spectral-workbench #88: comment posting says "There was an error" in popup, but still posts

jywarren, 2015-02-25:
http://spectralworkbench.org/analyze/spectrum/44204

[image]

jywarren, 2015-03-02:
This may be due to comment notifications; we should check that and write tests!

jywarren, 2015-03-02:
This doesn't happen on master on my local env but I also fixed the comment notifications in the refactoring I did, so that might've solved it.

btbonval, 2015-03-02:
`git checkout {part of a commit hash}` can check out the repo at any previous point in commit-time. You can always back track your code base to before you merged the changes into master.

btbonval, 2015-03-02:
... rather than using prod as the litmus test, that is ;)

jywarren, 2015-03-02:
Well, we had a scheduled push to production today anyways, and it wasn't going to get worse than it already was :-P

The push did not fix it. I'll dig a bit more.

jywarren, 2015-03-04:
Mailers hadn't been updated to Rails 3.2 conventions. Done now, working.

commit 44f75afe (2015-03-04): fixed mailers for ruby 3.2, close #88
closed 2015-03-04 by jywarren