publiclab/spectral-workbench #89: comments show html tags

jywarren, 2015-02-25:
[image]

(http://spectralworkbench.org/analyze/spectrum/44204) -- ignore the error in above image; just trying to show the <p> tags

btbonval, 2015-02-25:
FTFY
[image]

jywarren, 2015-02-25:
this should just be a matter of adding `raw` i.e. `<%= raw comment.body %>` because of new Rails 3 rules

jywarren, 2015-03-02:
Very odd -- this doesn't seem to happen in my local env. this line shows it may require the addition of `raw` in the template: https://github.com/publiclab/spectral-workbench/blob/master/app/views/comments/index.html.erb#L19

closed 2015-03-03 by jywarren