publiclab/spectral-workbench #98: can't select existing calibration (chrome & firefox, mac & windows)

mathewlippincott, 2015-03-03:
I can't assign an existing calibration to spectra in firefox or chrome on mac or windows. I can't seem to assign a calibration at time of capture or in the analyze window.

never seen this problem before...
[image]
[image]

jywarren, 2015-03-03:
Ah, this is probably a bug introduced in the new software. Will take a look
tomorrow afternoon :-(

jywarren, 2015-03-03:
OK, this is very odd -- i do see the buttons myself when logged in, but am wondering if somehow the table shown is pushed to the right off the edge of the window. Can you try selecting the text and dragging right, to see if there's content cut off? 

If that is correct, I can fix the column width.

mathewlippincott, 2015-03-05:
hmm.. that doesn't seem to do it.  I know I've lost buttons from small windows, but even on a big monitor I'm seeing the issue.

jywarren, 2015-03-05:
Plopped into a gist here: https://gist.github.com/gists

jywarren, 2015-03-05:
OK -- I think i identified the problem and fixed it in bc79269ca60e99a3313f05d7128b5100b749598b, 439b718254cf57e9066e826e80c30cab97bfe128, and 0899dbe35e61559a537f24f91c535cc12e6276f6

Can you test and reopen if it doesn't work?

mathewlippincott, 2015-03-12:
nope, still doesn't work.

mathewlippincott, 2015-03-12:
seems to work for Stevie, but I can't find a calibration button anywhere.

jywarren, 2015-03-12:
When I check, I don't see that the spectra in your screenshot are themselves calibrated -- are you sure they are, or that you've also searched for calibrated spectra? I think I can also make it only display calibrated spectra, which will make finding the most recent one easier.

jywarren, 2015-03-12:
I've made it so it only shows calibrated spectra. I don't know if that was the problem, but want to confirm that before continuing to investigate. Can you try?

mathewlippincott, 2015-03-12:
well, I can't calibrate my spectra so they aren't calibrated.... but whatever you did seemed to fix the issue.

jywarren, 2015-03-17:
I think the uncalibrated spectra were occupying the whole visible list, so removing them let you see and choose the most recent calibrated ones. 

Do you think this is solved completely now, or would you like to see pagination on the search results, so you can dig up very old calibrations? I'm going to close, but if you would like the latter, pls re-open.

jywarren, 2015-03-17:
And thanks for reporting!

closed 2015-03-17 by jywarren