publiclab/spectral-workbench #107: Gain-Correction

Stoft1, 2015-03-16:
After some experimentation ( http://publiclab.org/notes/stoft/02-25-2015/plab-spectrometer-gain-correction ) it looks possible that a default gain-correction curve could be included as part of SWB at least as an option. While there are significant limits to the PLab devices from the intensity vs. wavelength calibration, as a true calibration, a first-order approximation seems like a major improvement in the direction of providing users with more meaningful spectral results. As an adjunct to this, it might be helpful to add graphical markers at the wavelength limits -- i.e. ~400nm and ~700nm as beyond these limits the noise takes over as the camera has virtually no relative sensitivity.

jywarren, 2015-04-01:
Starting to explore pseudocode or stepwise procedure for ths sort of correction here: http://publiclab.org/notes/stoft/03-06-2015/plab-3-gain-correction#c11453

jywarren, 2016-02-01:
Going to use `$.get(url, function() {  });` using https://i.publiclab.org/system/images/photos/000/013/573/original/GCalPLab3Solux4700K_400-650nm.txt from https://publiclab.org/notes/stoft/03-06-2015/plab-3-gain-correction

jywarren, 2016-02-12:
This can now be run in v2.1.7; using:

````js
SpectralWorkbench.API.Core.useReference(false, spectrum);
graph.reload_and_refresh();
````

default is "/references/solux-4700K-stoft.csv" and there's also "/references/solux-4700K-rolloff-stoft.csv"

jywarren, 2016-02-18:
This is now working basically in v2.1.9 -- however we need a better UI.

jywarren, 2016-02-19:
1. OK, so I'm thinking -- we could display a list of webcam/lightbulb pairs for correction files, and also link to where you can do this process for an arbitrary pair and submit it to be added to the list. 
2. In terms of vocabulary, is "Gain calibration" the only way to refer to this? I want to be sure that if people are seeking "exposure calibration" it's clear whether or not this is the equivalent. 
3. We also need to display error. We could show an overlay of error, or a screenshotted graph of error, or we could simply name a maximum percentage, or we could do all three at different junctures.

jywarren, 2016-03-01:
Hi, Dave - sorry, i've been busy. Just wanted to get your input on "gain calibration" as terminology. Would "exposure calibration" be an equivalent term?