publiclab/spectral-workbench #108: Calibrated Wavelength Data

Stoft1, 2015-03-16:
I did two sets of measurments with PLab3.0 and, therefore, two separate calibrations. While the # of data points (and # of wavelength values) is the same, the wavelength values themselves are different ! OOPS! Yes, I'll admit they are close (just a few nm shifted) but for comparing supposidly calibrated spectra (wihtin SWB's present constructs) comparing two, calibrated (different cals) will have a residual offset error between the spectra. This was easily found from the downloaded 'raw' CVS files. There is no reason not to select a pre-defined set of calibration wavelengths for the data -- unless the horizontal resolution changes. IE. the displayed and downloaded data should not have a wavelength offset to the data sets -- but yes, two separate calibrations might have the CFL peaks at slightly different nm values..... but comparing the two data sets should not contain an 'X' axis shift to the data points.

jywarren, 2015-08-17:
Sorry, i'm not sure I follow, Dave - but does the new "range" setting potentially solve this? You could set all your range data to a given wavelength range?

However, setting a range does not remove excess data outside the range -- it just includes metadata to indicate where the range of "useful" data begins and ends. http://publiclab.org/notes/warren/08-06-2015/wavelength-range-limiting-in-spectral-workbench

But maybe I'm misunderstanding entirely?

jywarren, 2016-01-15:
Closing as I'm not sure what the next step is on this -- but please do re-open if you can help elaborate! Thanks!

closed 2016-01-15 by jywarren