publiclab/spectral-workbench #110: Calibration Points

Stoft1, 2015-03-16:
The calibration procedure is presently based only on the CFL Blue and Green peaks but because of distortions in the typical detected peak shape, it could be better to include the Red. Also, the procedure instructs the user to find the "highest" green peak. However, the Green peak is actually a double-peak and the highest is not always the "same" peak for all devices -- again, due to peak distortions and general device instability. The instructions should have the user pick the "right-most" peak -- ie. the one on the "red" side of the spectrum with the longer wavelength.

jywarren, 2015-08-17:
Can we split this issue in two, and make a recommendation to develop a macro for each? I'm changing this one to refer to the "green double wavelength" issue, but I believe the "include the Red" suggestion can be folded into the suggestion for a "nonlinear calibration macro" -- #55

jywarren, 2015-08-17:
For the double green peak, how should we do it -- what if we:

* change the procedure to find the highest two peaks within 10nm of the click, and 
* select the one with a shorter wavelength
* If there's only one, then just use that, assuming a lower resolution spectrum.

jywarren, 2015-09-16:
Addressing this in the new calibration system in #167; closing!

closed 2015-09-16 by jywarren