publiclab/spectral-workbench #113: Log in issue with smartphone

Slivias, 2015-03-19:
Hello. I have a serious problem with login. If i use pc everithing is ok using smartphone instead, i can log in publiclab.org but when switch to spectralworkbench i'm no more logged. If i try to login directly from spectralworkbench the site neither log in me nor give an error message. I tried chrome, firefox and opera.

jywarren, 2015-03-19:
Hi, I'm sorry you're having trouble -- what do you see instead of being
logged in? What is your username?

mikaelnelson, 2015-03-20:
This is happening to me on my desktop as well. On IE and Chrome.

closed 2015-03-20 by jywarren

Slivias, 2015-03-23:
Hi!
my username is sharky.
I try to explain each passage because can't do screenshot in this moment.
From spectralworkbench.org  pressing login is opened a pop-up window which request my users, i insert it and then press login button. This redirect me in the page where the site asking if I trust this site with my identity and  pressing Yes I am redirecting to spectralworkbench.org without to be logged in.

After ten times the log in was successful.

jywarren, 2015-03-23:
Did you see that I believe this was fixed last week? Can you try again and
confirm if it's still not working? Thanks!

Slivias, 2015-03-23:
Oh sorry!! I check right now with different browser and send the results!
sorry again for my distraction and thanks for your incredible work.

jywarren, 2015-03-23:
Thank you for your help!!!

Slivias, 2015-03-23:
Good news. Now works properly with Opera and Firefox (I am testing with
Android 4.4) but not with Chrome with after the request of login resend me
at the homepage of publiclab.
I don't know if it is a useful information.

jywarren, 2015-03-23:
Thanks, chrome desktop? What version, os? Thanks!

Slivias, 2015-03-23:
Chrome 41.0.2272.94 on Android 4.4.2

jywarren, 2015-03-23:
I wasn't able to reproduce on Chrome 41.0.2272.96 (i.e. it works for me); are you logged in to PublicLab.org on Chrome when you try it?

Slivias, 2015-03-24:
Good morning Jeffrey,
i go directly on spectralworkbench.org, the site require the username and
redirect me on publiclab.org/login, inserted again user and psw is open
publiclab.org without login.
But I think is a mobile problem because I have tried with another mobile, a
samsung galaxy trend lite with chrome 30.0.1599.82 ant it works.

I continue to try uploading my mobile and keep you update.

jywarren, 2015-03-24:
Hmm, OK - perhaps you could try in an Incognito tab on the original phone?

Slivias, 2015-03-27:
nothing,,,not even in incognito mode...at this point i wait the next
release of chrome and will try again.

jywarren, 2015-03-27:
That is so strange that it's just the one device. If you do have a chance
to test it on another phone with the same *version* of Chrome, that might
be helpful. My version was not identical, just very close.

commit 8961152b (2015-03-20): fixed login close #113