publiclab/spectral-workbench #119: Wizard for new users

jywarren, 2015-03-30:
* wizard: do you have a sample? 
  * Looks like this is a new map
    * http://mapknitter.org/maps/farmfield_2015
  * some thoughts on what the wizard asks?
  * interested in learning about features? ==> overview page
  * interested in reading what others have done?
  * need help? 
* subwizard: 
  * once click on "do you have a sample?" then followup questions; tree
  * other branches

@steviepubliclab - can you help flesh out what this "tree" looks like -- starting from the first time a user logs in? 

I imagine we'll show this only to users who have no spectra yet.

steviepubliclab, 2015-03-30:
Sure! 
[image]

steviepubliclab, 2015-03-30:
Sure: 
[image]

jywarren, 2016-01-11:
OK, my thought is to start with a display, which will replace the "recent research notes" on the SWB dashboard, to users with fewer than 5 spectra. This will have the following:

1. Hello! I see you are new to Spectral Workbench. Can we help you get started?
  * Learn about spectrometry
  * Analyze a sample I have
  * See what others have done
  * Browse Spectral Workbench features

Then this would go to the wiki. I know going to the wiki isn't maybe as nice as having pop-up menus in the page itself, but I'm putting lots and lots of links throughout the interface leading into the wiki with more info. How's this sounds, @steviepubliclab ?

steviepubliclab, 2016-01-19:
I like it. Any thoughts on walking someone through running a sample or compiling a set?

jywarren, 2016-01-20:
> walking someone through running a sample

We have [Taking and analyzing samples](http://publiclab.org/wiki/spectrometry-sampling) in the sidebar, and i'll link there, for sure, but that page needs work. Can we link into your workshops, or re-use content from there?

Compiling a set i'll add.

jywarren, 2016-01-20:
This can show on the dashboard, but I'm not sure what goes in each column; i can make a list of links, but we need pages for them to go to. Can you help? 

[image]

jywarren, 2016-02-01:
@steviepubliclab ping!

jywarren, 2016-02-08:
we can populate this with content from our new document!

steviepubliclab, 2016-02-29:
Ah sorry! I totally missed these pings! It could make it a little busy. A lot of links on the first step. Any way we could put the below options on other pages people can see once they click the main header?

jywarren, 2016-03-01:
Hi, Stevie -- no problem. I kind of need help composing this -- do you think you could do a mockup -- say, using multiple columns in a google spreadsheet, or something like that -- whatever's easiest? We can def. show one thing up front, then either expand as people click, or lead them to a different page.

jywarren, 2016-03-02:
We can use this eventually too: http://introjs.com/

steviepubliclab, 2016-03-03:
Here's a start. Not sure if all of these wikis are up to date. https://docs.google.com/spreadsheets/d/1kBdW9WWl1vjwjYbZJJZNnXCiOLaVo7HxdBviLOc7ZA8/edit#gid=0