publiclab/spectral-workbench #122: function the raw data in csv does not work

bosonbeard, 2015-05-03:
http://spectralworkbench.org/spectrums/49039
function the raw data in csv does not work 
button csv and csv raw give the same result

test on the browser Chrome and IE 9

jywarren, 2015-05-04:
Huh, you're right -- thank you for reporting. Are you looking for the JSON version (which is what's missing?)

jywarren, 2015-05-04:
Erg, sorry - my mistake. JSON is another button. Slow brain today. Will look into it!

bosonbeard, 2015-05-21:
I do not speak English but I can make a screenshot :)

see Figures 1 and 2 which made "csv" "csv (raw)"
values are the same

and none have the overall expansion, red, blue, green
as in the description:
CSV format description 
The CSV output is a 5 column comma separated text file. The first row contains text headers. Subsequent rows contain data in increasing wavelength order.

Column 1 contains the wavelength in nm.

Column 2 contains the average reflectance/transmittance/emissivity as a raw value integer in the range [0,255] as read. This value has not been converted to physical units and does not distinguish between reflectance and emissivity.

Columns 3, 4, and 5 are in the same format as column 2 and represent the red, green, and blue channels as read from the spectrum image.

[image]
[image]

jywarren, 2015-06-10:
I believe I've fixed this now. I'm sorry it took so long! Can you try it out?

closed 2015-06-10 by jywarren

bosonbeard, 2015-06-10:
Great! It works, thanks

commit c56db0ce (2015-06-10): json and csv export fixes #128 close #122