publiclab/spectral-workbench #125: calibration issue

Slivias, 2015-06-04:
Hello,
in spectralworkbench my user is sharky.
I have got a little issue. Analyzing some LED and using my calibration i have the right wavelength for the blue led (around 450nm) and wrong value for yellow and red. Then I have copied the calibration of user scottax2 obtaining good value for leds red and yellow, but wrong for blue one (shifted the wavelength around 600nm).
Also I don't understand why the spectrometer software works only if I use my mobile tited, in straight way doesn't acquire spectra 
in this way works: 
[image]

in this one no: 
[image]

Help!

thank u very much

Silvia

jywarren, 2015-06-08:
Unfortunately we can't lock the camera tilt sensor; it's too bad! 

As to your spectra, can you share a link to one? For calibration, you need to calibrate your own device; somebody else's data won't be aligned the same way. 

Sorry you're having trouble!

jywarren, 2015-09-16:
Closing as no response was found; please re-open if you see this and still have an issue! Thanks.

closed 2015-09-16 by jywarren