publiclab/spectral-workbench #126: camera recognition in Chrome

gretchengehrke, 2015-06-09:
Using Spectral Workbench in Chrome v43, in order for my computer to recognize the spectrometer camera, I need to clear my cache every single time (even if the only thing I have done since the last time is try to capture a spectrum), unplug the spectrometer, and plug it back in.  Even with that, the camera is not always detected, and I have not quite figured out the pattern of when it is and is not detected after clearing the cache.

jywarren, 2015-06-09:
Thanks for reporting; what operating system are you on?

gretchengehrke, 2015-06-09:
OS X 10.9.5

jywarren, 2015-06-24:
I know you can't yet debug this without a camera, but I was wondering if you could open your JavaScript console to see if there's an error related to the manifest cache. It'd look something like this:

[image]

jywarren, 2016-01-21:
Can you re-test this when you have a chance? Thanks!