publiclab/spectral-workbench #128: issue with json? Error using python

ronisegoly, 2015-06-10:
I am trying to load the json export using Python
http://spectralworkbench.org/spectrums/51827.json
I get an error
I asked for help in another forum and the answer I got is in the following link. i hope it helps
 https://gist.github.com/josuebrunel/bdc4ee50f1e1ea2d7651

jywarren, 2015-06-10:
Hi, Roni - I believe I've fixed this now. Can you re-extract your image and try again? Sorry for the trouble! 

As for the CSV - if you click the ? button there is a description of the CSV format. There was a problem with the Raw CSV which I've also now fixed.

closed 2015-06-10 by jywarren

ronisegoly, 2015-06-10:
works well, I parse it in python
My aim is to use it with Raspberry and other sensors such as Light sensor
to gauge the light level
cheers

*Roni Segoly *

*+972-54-6466264*

*
<http://s.wisestamp.com/links?url=tel%3A%252B972-54-6466264&sn=cm9uaS5zZWdvbHlAZ21haWwuY29t>*יוזמה
פרויקטים חברתיים וסביבתיים בע"מ <http://www.yozma.net/>

Yozma - social & environmental projects <http://www.yozma.net/>

Google+ Page [image: Google Plus Page]
<https://plus.google.com/u/0/+RoniSegoly/about>

* <http://en.wikipedia.org/wiki/Mohandas_Karamchand_Gandhi>*

ronisegoly, 2015-06-10:
> *+972-54-6466264 <%2B972-54-6466264>*

The python code

myrgb=[]
url = 'http://spectralworkbench.org/spectrums/52272.json'
response = requests.get(url)
data = json.loads(response.content)
dd=data['data']['lines']
for x in dd:

myrgb.append({'wavelength':x['wavelength'],'average':x['average'],'r':x['r'],'g':x['g'],'b':x['b']})

I create  new graph on Plotly as I will add data from other sensors
https://plot.ly/~rsegoly/526/spectrum/

*Roni Segoly *

*+972-54-6466264*

*
<http://s.wisestamp.com/links?url=tel%3A%252B972-54-6466264&sn=cm9uaS5zZWdvbHlAZ21haWwuY29t>*יוזמה
פרויקטים חברתיים וסביבתיים בע"מ <http://www.yozma.net/>

Yozma - social & environmental projects <http://www.yozma.net/>

Google+ Page [image: Google Plus Page]
<https://plus.google.com/u/0/+RoniSegoly/about>

* <http://en.wikipedia.org/wiki/Mohandas_Karamchand_Gandhi>*

jywarren, 2015-06-10:
Oh, very cool! Do you think there's a way to automate this, like could we
make an "export to Plot.ly" button?

ronisegoly, 2015-06-10:
in order to use plotly you have to install something on your computer and
create an account, so technically it can be automated (parse the data and
send to plotly) but you have to have an account there
I also use data.sparkfun to store data and that is also good place to
interface with, but same issue as above
You could of course offer ready to use python scripts
What do you think?

I have a question, any way I can data without he need to press buttons by
using URL for start capture, and download json?

*Roni Segoly *

*+972-54-6466264*

*
<http://s.wisestamp.com/links?url=tel%3A%252B972-54-6466264&sn=cm9uaS5zZWdvbHlAZ21haWwuY29t>*יוזמה
פרויקטים חברתיים וסביבתיים בע"מ <http://www.yozma.net/>

Yozma - social & environmental projects <http://www.yozma.net/>

Google+ Page [image: Google Plus Page]
<https://plus.google.com/u/0/+RoniSegoly/about>

* <http://en.wikipedia.org/wiki/Mohandas_Karamchand_Gandhi>*

jywarren, 2015-06-10:
To automatically capture, you'd have to interface with the JavaScript API:
http://publiclab.org/wiki/spectral-workbench-api

Hmm, I think we'll have to think harder about the Plot.ly thing -- but if
you're interested, please consider opening a new issue here in Github.
Thanks!

ronisegoly, 2015-06-11:
I got a bit lost with the API
I still do not have spectrometer (I ordered a kit which is on it's way, so
will know better once I use it
My setup will be a spectrometer on remote Raspberry pi and automation
So I will need to be able to...
1. get data from spectrometer on remote machine hopefully by running Python
scripts
2. Transfer the data to spectraworkbench to be analyzed or analyze locally
(there was offline setup?). This also needs to be done automatically
 3.Export using API so no need to browse and hit keys

Once I have the kit here I will dive in

*Roni Segoly *

*+972-54-6466264*

*
<http://s.wisestamp.com/links?url=tel%3A%252B972-54-6466264&sn=cm9uaS5zZWdvbHlAZ21haWwuY29t>*יוזמה
פרויקטים חברתיים וסביבתיים בע"מ <http://www.yozma.net/>

Yozma - social & environmental projects <http://www.yozma.net/>

Google+ Page [image: Google Plus Page]
<https://plus.google.com/u/0/+RoniSegoly/about>

* <http://en.wikipedia.org/wiki/Mohandas_Karamchand_Gandhi>*

commit c56db0ce (2015-06-10): json and csv export fixes #128 close #122