publiclab/spectral-workbench #130: assign anonymous authors to an "anonymous" user_id and optimize queries

jywarren, 2015-06-24:
* [ ] make a new user named "anonymous" with user_id = 0 (can we do this?) (in the migration)
* [ ] assign all anonymous spectra to this user (in the migration)
* [ ] remove any queries that include `WHERE author = 'anonymous'` and switch to `WHERE user_id != 0`
* [ ] consider removing the author field completely

jywarren, 2015-06-24:
This is odd -- when I run the suspect query (the main query for the front page of Spectral Workbench), it executes (in production on the live server) in ~230ms:

````ruby
> Spectrum.paginate(:order => "created_at DESC", :conditions => ["author != 'anonymous'"], :page => 1)
  Spectrum Load (232.3ms)  SELECT `spectrums`.* FROM `spectrums` WHERE (author != 'anonymous') ORDER BY created_at DESC LIMIT 30 OFFSET 0
   (0.2ms)  SELECT COUNT(*) FROM `spectrums` WHERE (author != 'anonymous')
````

jywarren, 2015-06-24:
OK, got it:

````ruby
Spectrum Load (51559.0ms)  SELECT `spectrums`.* FROM `spectrums` WHERE `spectrums`.`author` = 'anonymous' AND (user_id != 0) LIMIT 20
````

jywarren, 2015-06-24:
Almost ready; migration runs, tests pass. Dealing with test warnings related to i18n

btbonval, 2015-06-24:
Keep in mind the problem with response might not be related to that query
at all. In fact, I think you've provided evidence that it might not be. If
it were caching or something, you'd think the first time would go slow and
then the next would go fast. But you posted the opposite behavior. Though
to be fair the queries were different.

Anyway this should be fixed regardless of what is causing slowdowns.

jywarren, 2015-06-24:
Agree should be fixed either way. At least we'll eliminate a possibility as well!

jywarren, 2015-06-24:
And it's in production. Watching the Rackspace monitoring graph!

jywarren, 2015-06-24:
Closing, deferring the complete purge of "author" field for now.

closed 2015-06-24 by jywarren