publiclab/spectral-workbench #136: Spectral workbench not saving spectrum

anks123, 2015-07-09:
Hello,
I am unable to save any spectrum on the spectral workbench site.
After taking the live capture, I added the details and then hit save.
However it took me to a page which said "We're sorry, but something went wrong"
This problem occured post 12:30 pm IST.
What do I do ?
My username is ankur_bodhe

jywarren, 2015-07-09:
I tested this and it seemed it was only spectra where we specify a calibration source. I believe I've fixed the problem and am pushing the fix live -- can you check it out and tell us if it worked?

jywarren, 2015-07-09:
And re-open this if it did not, please! Thanks again.

closed 2015-07-09 by jywarren

anks123, 2015-07-09:
Hi, 
I just uploaded a spectrum and it works 
Thanks a lot