publiclab/spectral-workbench #140: Spectral workbench spectrum CSV file data error

anks123, 2015-07-15:
Hey 

So I tried downloading the CSV file today for a spectrum and once the file is downloaded and loaded onto a spreadsheet, the wavelengths show up to be 0 for everything.
anything that can be done ?

jywarren, 2015-07-15:
You're right - will try to work on that today. Sorry for the bug! Are you
interested in contributing code, at all? Thanks!

jywarren, 2015-07-15:
Fixed! Thanks again.

closed 2015-07-15 by jywarren

anks123, 2015-07-15:
Thanks! well I'm not an expert coder, I'd try though

jywarren, 2015-07-15:
Hey, every thing helps. The first step is getting Spectral Workbench
installed; have you tried that?

anks123, 2015-07-21:
No havent tried that yet .. where can I install it from ?

jywarren, 2015-07-21:
There are instructions at https://github.com/publiclab/spectral-workbench
-- and please reach out on the developers list (
http://publiclab.org/wiki/developers) if you need some help getting it
started! Thanks.