publiclab/spectral-workbench #146: Errorpage when saving spectrum with existing calibration

YogiBGitHub, 2015-08-11:
The "We're sorry, but something went wrong" page appears when trying to calibrate a spectrum with an existing calibration. But when going back to the actual spectrum, it seems to have been calibrated any way. Also, a new calibration does not appear in the list when choosing to use an existing calibration. I have also noticed that when smoothing a spectrum ten lines below, the spectrum is not saved. So - I guess there is an issue with the filesystem... or something like that.
Bjorn

jywarren, 2015-08-11:
Thanks a lot - I have a fix for this which I hope to post shortly.

jywarren, 2015-08-11:
Hi, I pushed the new changes - can you check that this works now?

YogiBGitHub, 2015-08-11:
Thanks a lot. Now it works very well again.

closed 2015-08-18 by jywarren