publiclab/spectral-workbench #190: Database load on query for popular spectrums

icarito, 2015-09-03:
Following is an excerpt from the logs in production server at spectralworkbench.org.
We are experiencing mini-outages that can be reproduced when browsing the `/popular/*` pages.

I've checked the source at https://github.com/publiclab/spectral-workbench/blob/d2ab8917170ee13552804903158aab68e78e77b5/app/controllers/likes_controller.rb and I think one solution is to make `"likes_count"` an indexed field, currently it isn't.  
```
Started GET "/spectra/show/1107" for 91.121.211.59 at 2015-09-03 02:30:21 +0000 
Started GET "/sets/remove/520?s=10055" for 188.165.15.64 at 2015-09-03 02:30:21 +0000 
Processing by SetsController#remove as */* 
  Parameters: {"s"=>"10055", "id"=>"520"} 
Redirected to http://spectralworkbench.org/login?back_to=/sets/remove/520 
Filter chain halted as :require_login rendered or redirected 
Completed 302 Found in 0.9ms (ActiveRecord: 0.0ms) 
Started GET "/popular/recent?page=1482" for 179.7.97.153 at 2015-09-03 02:30:21 +0000 
Processing by LikesController#recent as HTML 
  Parameters: {"page"=>"1482"} 
  Rendered spectrums/_list.html.erb (569591.4ms) 
  Rendered likes/index.html.erb within layouts/application (569602.8ms) 
  Rendered layouts/_locals.js.erb (0.5ms) 
  Rendered sessions/_login.html.erb (0.7ms) 
  Rendered layouts/_header.html.erb (1.3ms) 
  Rendered spectrums/_search.html.erb (0.4ms) 
Completed 200 OK in 569630.6ms (Views: 143.5ms | ActiveRecord: 569484.3ms) 
Started GET "/spectrums/1107" for 91.121.211.59 at 2015-09-03 02:39:46 +0000 
Processing by SpectrumsController#show as HTML 
  Parameters: {"id"=>"1107"} 
  Rendered spectrums/_list.html.erb (565170.8ms) 
  Rendered spectrums/_list.html.erb (594617.8ms) 
  Rendered likes/index.html.erb within layouts/application (565178.0ms) 
  Rendered layouts/_locals.js.erb (0.0ms) 
  Rendered sessions/_login.html.erb (0.2ms) 
  Rendered layouts/_header.html.erb (0.1ms) 
  Rendered spectrums/_search.html.erb (0.0ms) 
Completed 200 OK in 565183.2ms (Views: 60.7ms | ActiveRecord: 565121.0ms) 
  Rendered likes/index.html.erb within layouts/application (594622.8ms) 
Started GET "/" for 179.7.97.153 at 2015-09-03 02:39:46 +0000 
  Rendered layouts/_locals.js.erb (0.0ms) 
  Rendered sessions/_login.html.erb (0.2ms) 
  Rendered layouts/_header.html.erb (0.1ms) 
Processing by SpectrumsController#index as HTML 
  Rendered spectrums/_search.html.erb (0.0ms) 
Completed 200 OK in 594627.2ms (Views: 60.9ms | ActiveRecord: 594565.7ms) 
Started GET "/match/search/21981" for 66.249.64.201 at 2015-09-03 02:39:46 +0000 
Processing by MatchController#search as HTML 
  Parameters: {"id"=>"21981"} 
Redirected to http://spectralworkbench.org/spectrums/21981 
Completed 302 Found in 14.7ms (ActiveRecord: 13.2ms) 
Started GET "/spectra/rotate/22855" for 188.165.15.64 at 2015-09-03 02:39:46 +0000 
Processing by SpectrumsController#rotate as */* 
  Parameters: {"id"=>"22855"} 
Redirected to http://spectralworkbench.org/spectrums/22855 
Completed 302 Found in 1.9ms (ActiveRecord: 0.7ms)
```

jywarren, 2015-09-03:
What about indexing on `created_at`? since that's what it's sorting by? 

        @spectrums = Spectrum.order('created_at DESC').where('user_id != 0').page(params[:page])

I just created a pull request for this change but we need to hear from @Sreyanth about his changes -- he's asked that we hold off on a merge until he's ready with his rebased branch.

icarito, 2015-09-04:
I was guessing from looking at the code, there are two queries I saw:

      @spectrums = Spectrum.order('like_count DESC').where('user_id != 0').paginate(:page => params[:page], :per_page => 24)

It is worth testing.

jywarren, 2015-09-09:
This was just pushed to production. Let's see how it does! @icarito  please check in after a day or two to see if it had any effect.

closed 2015-09-09 by jywarren

jywarren, 2015-09-11:
I narrowed this query after looking into the full MySQL EXPLAIN; in 4fdd92e4459d23fd2930f6168e174aef3259d759