publiclab/spectral-workbench #192: ability to freeze/snapshot a spectrum by cloning it

jywarren, 2015-09-03:
do this as lazily as possible -- if possible, not on `subtract:9` tag creation, but only if there is a reference to the current spectrum that'd be disturbed?

Stoft1, 2015-09-03:
Jeff, I'd suggest keeping data separation between users (one user can never operate on or with the original data of another user -- it must always be on/with a local copy of that data) separate from the data history mechanisms available to a user.

Eg: User-A saves a spectrum. Filters of that spectrum only exist as a list of operations to be performed before display or download but a particular set of filters can be saved as a new version (just not the modified data itself). User-B can reference the original or a filtered version, but if they want to actually use that data, they must make a copy. If they want a filtered copy, they get the original data along with the mod list for the version they wanted to use. Since they now have a copy (which has appropriate tags as to its providence for future reference) User-B's work with that data is now independent from User-A data so no residual dependencies cause backward effects on A.

So, 1) original data is protected as you said, 2) no need to save the data of filter operations (just what filter to use), 3) another user must use a local copy (with providence and filter list) and 4) original data is never affected by anything from another user.

Hope I've not missed your points.

jywarren, 2015-09-03:
Great - actually inter-user rules are somewhat simpler to think about, and I tend to agree with you -- but do you mean that if you subtract someone else's spectrum (Spectrum A) from your own, you must make your own copy of A, say, A1, owned by yourself, before you can do that? (we might automate it, but still?)

Stoft1, 2015-09-03:
There's actually two cases of that. 1) If B subtract's A's original, and 2) if B subtracts A's filtered data. Part of this is the question of 1) is it GUARANTEED that A's original data will ALWAYS exist and NEVER be changed/updated/replaced. If so, then B is safe to use that data -- but if B uses filtered data, then that filter might be changed by A. See the issue? To avoid all these potential dependencies, it is much much easier to just say that B is required (must) grab an make a local-to-B copy of A's data, along with the flags for filtering. Then, B is free to subtract or whatever w/o any such dependencies. This way, B is free to then delete (but not edit) a copy of A's original data if/when it is no longer useful, w/o any effect on A. Now IF you could actually guarantee that ALL of A's original data is permanent and unalterable, then the copy is not needed --- except (there's always at least one of these) if B doesn't want the original but a specific filtered version. B would then use A-original+filters and and subtract .... and then A would change a filter and B's results would change. Ah, so you say no, the A-data is unchanged but B decides on the filters -- which could work, but I suspect this sequence would be much more confusing for user B to keep track of -- especially if there were multiple filters on the A data they wanted and didn't try to keep track of everything that A did to get the spectrum display that B wanted to use. So, I'm suggesting the A-B separation and that B always makes a copy of A+filters as being 1) simpler and 2) eliminates a lot of potential dependencies. Dependencies are very difficult to program or program-around.

jywarren, 2015-09-03:
Unfortunately there are two scenarios where there could be a change to A:

1. The extraction of numerical data from the image is itself a kind of
filter, although the original image never changes. But snapshots would fix
that, so we're good.
2. The original data could be deleted. It's hard to say if we could come up
with a system where snapshots could outlive their parent spectra. It seems
dubious. But do we forbid deletion of any data? Or any data which is
referenced anywhere?

Stoft1, 2015-09-03:
I'd suggest that, given a substantial amount of existing original data may be eventually need to be sent to the bit bucket as just too old, a fluke, can't recall what it represents, etc, basing any scheme on assumed permanence is likely to fail. Also, any data extracted from a spectrum may also be subject to change as even calculations, corrections, calibrations etc tend to evolve and improve with time. While raw data can easily be flawed (eg. a config error) and filtering can change, at least the raw data is still the measurement source (while it still exists) and single-point-source is generally the best point to start.

Now, under the scenario of User-B making a copy and then data-A being deleted, it might not be too difficult to then tag B's copy to show that the original A-data is changed/deleted. Probably don't want the overhead of always doing this, but could, perhaps, perform the check if user B clicks on a request to check.

Stoft1, 2015-09-04:
There is one case for 'permanent' data I can think of -- reference data which has met some standard as a reference which most may not have the ability to produce -- as opposed to the bulk of contributed data which has not met (thought it later could) some standard. The 'reference' data might be saved as 'permanent' and the general contributed data would be potentially volatile. Of course, this just again makes the case for a data handling process where it must be assumed the raw data might go away -- so create a local copy in order to use it.

jywarren, 2015-09-11:
I'm starting to think in parallel about how to represent this to the user... frozen/snapshotted data (or whatever we ultimately decide upon) could be shown in the new "operations" table, at the point where it was snapshotted. Any later operations would appear after it. 

[image]

This could be a vote for the mental model where to remove operations that would affect a snapshot, you have to fork off before that point -- which could also be shown in the operations table with a forking symbol: 

[image]

jywarren, 2015-12-01:
OK, circling back, i'm going to start reorganizing the top of this issue and implementing it. My original thought was that we'd want to avoid making an extra copy whenever possible, but now I'm tending to think that a really predictable, consistent convention will be worth the redundancy. So, I like Dave's suggestion that any time you reference something, say with a powertag like a subtraction, we auto-generate a snapshot, owned by whomever referred to it, so they maintain control over that frozen copy from that moment in the data's history. There might be multiple snapshots from that moment -- identical, but connected to different referring spectra. Redundant, but an easy model to follow.

One other question, then -- if someone else snapshots your spectrum, then you want to delete your data, are you allowed to? By the terms of the licensing, we don't have to permit users to delete their data, since they've CC-zero licensed it. We could allow deletion UNLESS there's a snapshot relying on the data.

jywarren, 2016-01-04:
This is complete. Whoa.

closed 2016-01-04 by jywarren