publiclab/spectral-workbench #203: Unable to Delete Images from Profile

gamewizzzz, 2015-09-29:
When I try to view some images I've uploaded, I get an error:
"We're sorry, but something went wrong.
You may have mistyped the address or the page may have moved. You can try searching for spectra or sets here:"

I've tried to delete the corrupt images, but I think the only way to delete an image is to view it first, but I get the error every time I try to view them.

Effected images:
http://spectralworkbench.org/spectrums/57827
http://spectralworkbench.org/spectrums/57826
http://spectralworkbench.org/spectrums/57825

Any assistance is appreciated.

jywarren, 2015-10-02:
Hi, sorry for the slow reply; I also see the error, but they are available at the new beta interface: http://spectralworkbench.org/spectrums/show2/57827

For each URL, just add the extra "/show2" and you should be able to delete them from there. 

Again, apologies for the trouble! Do you recall what you did leading up to their not displaying, so we could try to find the original issue?

Thank you for reporting!

gamewizzzz, 2015-10-03:
Adding in the extra "/show2" does the trick, thank you.

Memory is a little fuzzy since at the time I was learning to use the spectral workbench interface with the desktop v3 color meter. The only things I can remember doing differently on the corrupt images were giving them a long name, saving them with no calibration, and maybe logging off spectral workbench before the images finished uploading.

The first 3 corrupt images are un-calibrated, and seem to have the same name (IMG_2146_%28CFL_desk_light%29.JPG). The 4th corrupt image is calibrated and has a short name, but it has no description and I think I waited 2 minutes for it to upload before I exited the browser.

I will leave the corrupted images up for another few weeks in case you want to analyze them some more.

jywarren, 2015-10-05:
Thanks -- I suspect that the filenames including some odd characters may have caused the problem, and am labeling this issue accordingly. Hopefully we swing back and fix this soon.