publiclab/spectral-workbench #213: Do not copy calibration.

bosonbeard, 2015-10-13:
I select any. to calibrate the spectrum, and then writes
You successfully cloned Spectrum # 59HHH, but the scale does not change and the range is not calibrated wavelength (everything stays the same)
http://spectralworkbench.org/spectrums/59207

jywarren, 2015-10-13:
Hi, where did you click that clone button? We are adjusting the language
for "clone" -- it now means to copy an entire spectrum, whereas "copying
calibration" means just copying the calibration, not making a complete copy
of the spectrum.

Jeff

bosonbeard, 2015-10-13:
Hello, thanks for answer.
I make a screenshot
https://habrastorage.org/files/a5e/72a/63e/a5e72a63eb024340b5f861d9f74073af.jpg

jywarren, 2015-10-13:
Ah, thank you. I'm submitting a fix now. Closing -- please re-open if this does not work.

closed 2015-10-13 by jywarren