publiclab/spectral-workbench #214: incorrect display of the spectrum after calibration

bosonbeard, 2015-10-13:
after calibration of the spectrum image is automatically reflected in an incorrect state (auto flip image)
http://spectralworkbench.org/spectrums/59226

jywarren, 2015-10-13:
1. Can you share what the original image was?
2. Was it uploaded or live-captured?
3. What URL or interface was it calibrated from? Were you using the beta
preview interface or the standard one?
4. It sounds like it was not initially backwards, but was flipped after
calibrating -- do I have that right?

Thanks for reporting!

bosonbeard, 2015-10-13:
1. https://habrastorage.org/files/9cc/696/3f9/9cc6963f97944517b9657d5a9e0506a5.jpg
2. upload
3 I do not understand 100% the question, I think the standard interface
4 yes

thanks for the answer

jywarren, 2015-10-13:
OK, so under "More Tools" you can reverse the spectrum again. Sorry - this bug is known (#6) and we'll be addressing it in the upcoming SpectralWorkbench 2.0 release. I'm going to close this here for now.

closed 2015-10-13 by jywarren