publiclab/spectral-workbench #218: Feature Request: Export Spectral Data in Single nm steps (i.e., 340, 341, 342)

JJCone, 2015-10-27:
I'm trying to convert the spectral data to cone receptor activation in the retina. The cone fundamentals are rated in single nm steps but the spectral data (when export as CSV) comes out in weird step sizes.

jywarren, 2015-11-04:
It currently comes out with one datum per pixel. I'm not sure what's the
best way to do your request -- should it be an export option, or perhaps
some kind of API macro? Currently the SVG export is defined in the API
itself, in
https://github.com/publiclab/spectral-workbench/blob/master/app/assets/javascripts/spectralworkbench/SpectralWorkbench.API.Core.js#L54

Perhaps we could move that into the Spectrum model and add another for your
type of export?

JJCone, 2015-11-04:
Thanks for your response

I didn't realize the data was created by measuring across pixels, but that
makes sense since you are spreading the light across the surface of the
webcam. Perhaps then the easiest way to get the data into nM sized steps
would be to create a rough estimate of the intensity at each nM of
wavelength by taking a weighted average of nearby pixels.

While not entirely accurate, I did come up with one workaround that worked
out okay for teaching purposes. I exported the data as CSV and dumped it in
MATLAB. I then down sampled the cone fundamentals (which range from 400 to
700 nM) a little bit so the resolution would more closely match the
resolution of the CSV file across the same wavelength range. At that point
you just take the dot product of the spectral intensity by the Cone
fundamentals to end up with a number that corresponds (approximately) to
the activation of each of the Cones.

The one thought that comes up here is that it would be handy to be able to
select the range of wavelengths you wanted in your CSV file prior to
export.

Thanks again

jywarren, 2015-11-04:
Glad you worked it out -- in the API, the relevant method would be:

[spectrum.getIntensity(wavelength, channel)](https://github.com/publiclab/spectral-workbench/blob/master/app/assets/javascripts/spectralworkbench/SpectralWorkbench.Spectrum.js#L46) - which finds the **nearest** datum in the given channel

On the other hand, if you wanted to **average** to find a more precise measure which would not be an actual recorded data point, you could use something like the approach in [spectrum.calibrate()](https://github.com/publiclab/spectral-workbench/blob/master/app/assets/javascripts/spectralworkbench/SpectralWorkbench.Spectrum.js#L246), which interpolates.