publiclab/spectral-workbench #219: firefox login problem

steviepubliclab, 2015-10-28:
I'm getting a bug using SWB on firefox, mainly seems like a login issue. 

Every time I try to login - either from the direct Log In button, or Capture Spectra--> Begin Capture --> Save (when I do this the login page pops up in a new tab)  -- I get the 499b error. The page seems to pick one of my logins at that point - the dominant one used on this computer - but then continues to give me new popups for logging in with the same error each time I try to capture spectra. 

I noticed this login is different when I use SWB on my chrome browser. There I get the login prompt as a popup on the same browser tab and that seems to work fine.

jywarren, 2016-02-08:
1. Can you try doing this in Firefox in Private mode, to see if it has to do with any cached data?
2. If that works, can you log out of PublicLab.org and SpectralWorkbench.org in regular mode and try again?
3. Whether or not #2 works, can you clear your cache in Firefox and try once more?

Thanks, if you could report back with what happens in each case, esp. what you see in the header re: logged-in status and username on each site, that'd be very helpful.

jywarren, 2016-04-04:
Closing this, but if you see it recur, please re-open and try to go through the debugging steps mentioned. Thanks!

closed 2016-04-04 by jywarren