publiclab/spectral-workbench #232: can't login

bosonbeard, 2015-11-15:
I can not log into your profile and upload a photo of the spectrum.

I enter the password, I see window 1 (in the screenshot)
click NUY (screenshot 2), but after the transition to following page, I am not in the system.
I see snovo inscription Username instead of my nickname

[image]

bosonbeard, 2015-11-15:
Use UIE 11 and Chromium (iron) 43

bosonbeard, 2015-11-15:
IE 11

YogiBGitHub, 2015-11-15:
I am having the same problem - again. A couple of days ago I tried to delete cockies and tempfiles in the browser, and then it suddenly worked. But now the login does not work any more, no matter what I do, Chrome, Firefox, IE 11 or Edge.

jywarren, 2015-11-15:
I'm sorry for the issues - we (@icarito and i) just forced SSL security for
publiclab.org and turned it on optionally for spectralwb.

I thought I'd made the necessary changes but i suspect somethings wrong
with them - I will be able to debug first thing tomorrow - apologies!

jywarren, 2015-11-15:
Ah - on this line, don't know how I missed it: https://github.com/publiclab/spectral-workbench/blob/master/app/controllers/sessions_controller.rb

jywarren, 2015-11-15:
Needs to be https

jywarren, 2015-11-15:
OK - i've initiated a database migration to fix this. It should be resolved
in about 30 minutes -- can you test then and tell me if it worked?

THanks!

jywarren, 2015-11-15:
Should be complete. If it works, can you please close this issue? Thanks!

YogiBGitHub, 2015-11-15:
Many thanks Jeffrey! Now login works like a dream.

YogiBGitHub, 2015-11-15:
Since I didn’t open this issue I cannot close it.

jywarren, 2015-11-15:
@bosonbeard can hopefully close it if the error is resolved for him too.

bosonbeard, 2015-11-15:
yes, thanks

closed 2015-11-15 by bosonbeard