publiclab/spectral-workbench #236: Cannot view uploaded spectrum

Starryblack, 2015-11-27:
Hi,

When i try to upload an image (100kb .jpeg) Spectral workbench always leads me to the Sorry something went wrong page. The spectrum appears in my dashboard but pressing it leads to that page. Is there possible a fix for this? Thanks.

username is wms 
[image]
using chrome
issue: https://spectralworkbench.org/spectrums/63638

jywarren, 2015-11-30:
Hi, can you take a picture of the form you use to submit this image, so we
can see the title & metadata before you upload? This will help us debug.
Thanks for reporting!

jywarren, 2015-12-18:
Please feel free to re-open, but as I haven't heard back, I'm going to close this up. Thanks!

closed 2015-12-18 by jywarren