publiclab/spectral-workbench #238: Make "find lambda max" function

gretchengehrke, 2015-12-15:
Hi Jeff, 

The most common way of distinguishing different spectra is from their lambda max (wavelength at which they have the highest fluorescence intensity). That will be a more useful function than "find graph center".  Would it be possible to add this function, and hopefully fairly quickly (so we can use it in analysis of the OTK beta tests in January)? 

Thanks!
GG

jywarren, 2015-12-15:
Hi, Gretchen - it's easy to add the function, but without intensity
calibration, the highest peak read will be affected by the differing camera
sensitivity for different wavelengths, so comparisons will be limited to
same-camera-only. There is also an issue that there are "dips" between
color channels, where the camera is less sensitive. This is why we settled
on overall graph center, which is also affected, but we hope, less so. you
can read about this direction here:

https://publiclab.org/notes/warren/07-30-2014/equalizing-area-of-spectral-graphs-for-comparison
https://publiclab.org/notes/warren/09-23-2014/better-smoothing-and-equalizing-graph-height-for-comparison-of-oil-spectra

Do you think that given the camera sensitivity curve limitations, you still
want highest intensity? I'm of course happy to make it for general use. Let
me think about how to do it most simply.

gretchengehrke, 2015-12-15:
I think it will still be a useful tool, but that is a really good point about the "dips." Do you know if cameras tend to have different sensitivities for different wavelengths within the green and within the red and within the blue sections? I'll read your research notes and read up on that issue. If they are pretty consistent over the range of each R, G, or B zone, then perhaps a lambda max of each... If not, yes, this could be an issue.

jywarren, 2015-12-15:
> different sensitivities for different wavelengths within the green and
within the red and within the blue sections

Do you mean, if we only take one channel? Yes, they are going to be most
sensitive at the middle of the color band. I do think you could find maxima
per channel, but what would you then do, average the three maxima? Or what
would be your methodology for three maxima instead of one?

gretchengehrke, 2015-12-15:
Hi Jeff, 

Do you have any idea how much that sensitivity would vary across the range of wavelengths of a given band?  What we care about here is not actually the intensity -- we care about the wavelength. We care about the wavelength at which the highest intensity occurs, so a problem would arises when the different sensitivities would obscure our ability to identify which wavelength actually as the highest intensity fluorescence.  For example, if we had a compound that most strongly fluoresced at 540 nm, but our webcam was more sensitive at 520 nm than at 540 nm, so even though 540 nm is the true peak, our webcam spectra might actually show a higher intensity at 520 nm, we would mistakenly assign the lambda max at 520 nm, and could quite possibly mis-categorize the compound because of that. Does that make sense?

jywarren, 2015-12-15:
Yes, exactly - so, for a real-world example, look at this black body
spectrum of the sun: https://spectralworkbench.org/spectrums/show2/49737

It should be a smooth curve, but you can see how much each channel shows.
This would almost certainly affect the max peak position if it fell between
two color channels.

gretchengehrke, 2015-12-15:
Wow -- That is a great example. It should actually even be an asymmetric smooth curve, steep on the UV-V side and an extended tail on the R-IR end. Hmmmm... This complicates things. 

Though it would not be useful for external comparisons, for internal comparisons among spectrometers  using the same type of webcam, do you think a lambda max for each channel would be useful?

jywarren, 2015-12-15:
Sure, I'd bet our webcams would be reasonably consistent. It's also easy to
test if Liz or stevie or someone else who has multiple can try it out.
Pointing several at a white daylit wall, say, same angle.

jywarren, 2016-02-08:
Where would this tool report the result? What about next to the operation in the Operations listing to the lower right of the spectrum?

gretchengehrke, 2016-02-08:
Yeah, that would be great

jywarren, 2016-02-09:
Would something like this work? 

[image]

jywarren, 2016-02-09:
We could also have an "Output" column, with more explanation. 

Note to self: I achieved the above with `$('#tag_738 td.title').append('<span class="badge badge-success">Result: 45</span>')`

gretchengehrke, 2016-02-09:
I don't think I understand what this is saying... What is 45? What is being forked? I'm confused. I just think there should be a button that says "Find Lambda-max" and then there is an output saying "Lambda-max = 535 nm" or something like that.

jywarren, 2016-02-09:
I'm talking about making it save the calculation and its output so others
can see it.

gretchengehrke, 2016-02-09:
Oh, yes, I think that it will be most useful if others can see it and there can be a record of it. Sorry I still don't understand where it is showing what the lambda max is though.

jywarren, 2016-02-09:
Sorry, I didn't mock it up entirely; it'd have an operation called
"calculate:lambdaMax" or something, and the output would be put in the
green oval.