publiclab/spectral-workbench #268: Live-Capture Not Working

zvinitski, 2016-01-22:
When I go to live capture it says that the site may have been moved or removed. I use 3 different internet browsers (Safari, Firefox, Chrome) and it say the same thing.

jywarren, 2016-01-22:
Thanks, were tracking this and should have a solution later tonight.

jywarren, 2016-01-22:
Working on this now; I see an error for /capture/offline:

````
Started GET "/capture/offline" for 96.224.88.136 at 2016-01-22 03:03:24 +0000
Processing by CaptureController#offline as HTML
Completed 500 Internal Server Error in 3.8ms

ActiveModel::MissingAttributeError (missing attribute: data):
  app/models/spectrum.rb:218:in `clean_json'
  app/models/spectrum.rb:516:in `latest_json_data'
  app/models/spectrum.rb:418:in `wavelength_range'
  app/controllers/capture_controller.rb:32:in `offline'
````

jywarren, 2016-01-22:
OK, I believe this is resolved as of #267. Can you confirm?

closed 2016-01-22 by jywarren

ebarry, 2016-01-22:
I still see this error: 
[image]

ebarry, 2016-01-22:
Also seeing the "We're sorry, but something went wrong" in Firefox:
[image]

jywarren, 2016-01-22:
Additional fix just went in. Can you test please?

ebarry, 2016-01-22:
The capture interface works great in Chrome! 
Please note that I did not yet go through saving a spectrum and testing all the features. 

I do have to report this strange muppet sighting: 
[image]