publiclab/spectral-workbench #273: Compare with another spectrum - not mine

YogiBGitHub, 2016-01-22:
When I am trying to compare with another spectrum, selecting "Your spectra", I get every others spectra, so It's really not usable. Must be some filtering error.

steviepubliclab, 2016-01-23:
Also have this problem. I can't seem to find any of mine.

jywarren, 2016-01-23:
We should fix this, but I think you can search for the ID # to find
spectra. See if that works?

jywarren, 2016-01-23:
Maybe "your spectra" menu is not hooked up right.

YogiBGitHub, 2016-01-23:
Can’t find the ID# either… but the name I have saved it with works.

steviepubliclab, 2016-01-23:
I can search for a spectra I named and find it that way. I can't seem to
find on my spectra scan where the individual number is.

jywarren, 2016-01-25:
You can use the # from the URL. But I'm adding this to outstanding bugs to fix for 2.0, thanks.

closed 2016-01-27 by jywarren

jywarren, 2016-01-28:
I believe this is resolved now -- can you give it a try?

commit 53519fea (2016-01-27): close #288 multiple /choose/ queries from SpectraPane, close #273 searching for Your Spectra in SpectraPane