publiclab/spectral-workbench #274: Delete spectra and equalize height

YogiBGitHub, 2016-01-22:
In the new version I miss simple things like deleting a spectra and equalizing height (when comparing).

jywarren, 2016-01-22:
Hello, thanks for reporting; on which page?

YogiBGitHub, 2016-01-22:
Here is the url for the spectrum I try to delete:

https://spectralworkbench.org/spectrums/show2/68729

Maybe I start to get blind, but I cannot find a way to delete a spectrum.

jywarren, 2016-01-25:
It should be on the left next to the "Like" button. But if you can't see it, please upload a screenshot and re-open this issue. Thanks!

closed 2016-01-25 by jywarren

YogiBGitHub, 2016-01-25:
Yes, I can see there is a delete icon, but for some spectra it is disabled.

jywarren, 2016-01-25:
If you hover over it, does it tell you why?

YogiBGitHub, 2016-01-25:
No, when disabled, it does not tell anything.

jywarren, 2016-01-25:
OK, we'll try to address this in the next release -- thanks! Some are
disabled because they are being used by other spectrums' operations.

YogiBGitHub, 2016-01-25:
Thanks. I guessed there were some dependencies behind.

I also miss the “equalize height” feature. I use this to make a relative output comparison. In each measurement of some red leds I use stronger blue leds as a reference. So when comparing several spectra, I use equalize height to level all the blue reading. Then the red ones can be compared relatively speaking. Please see the attached picture, where the blue peaks are cut away. It has been very useful.