publiclab/spectral-workbench #285: Capture glitch with Firefox 43

Stoft1, 2016-01-26:
These are likely related to each other.

1 - I start-up SWB, the camera works, the graphic image displays with color bands, I can set the pixel-line and then click the Capture mode button. The spectral trace appears but only shows active data on the left half (up to about 530nm) -- the right half is "dead"; no data. Further, there are no signals corresponding to any spectral peaks despite a number of CFL signals.

2 - Assuming this glitch is easy to fix, I also note that at the above point in capture, SWB doesn't know anything about wavelength because I've not told it anything about any CFL reference for my device. So, why is the X scale marked in (nm) instead of, perhaps, just showing the pixel count. Showing (nm), when that is not known, is false.

[image]

[image]

[image]

jywarren, 2016-02-10:
Hmm, I'm wondering if somehow it's gotten stuck in a "rotated" state. Can you post a screenshot of the entire page? And, separately, can you try it in "private mode" to be sure it's not caching anything?

Thanks, Dave!

jywarren, 2016-02-12:
via @Stoft1 - 

> 1) Video w/spectrum colors do appear in small window but on clicking 'Capture' no signals appear and display of data is either left-shifted or cut off at 520nm. Seems an old Cal is being used (improperly) for this display. This bug was previously reported -- but still exists in this new relase of SWB.

> 2) In attempting to 'reset' this issue, the 'New Calibration' option was selected and 'Switch cal' was triggered -- only to get an error page with a link to report the issue.

Thanks, Dave - #2 should be broken out into its own issue. I'm still trying to reproduce #1; can you try it in Private mode please so we can check if it's related to older cached code or data?

jywarren, 2016-02-12:
Dave also added:

> I cleared cache with no effect. 

Can you tell me which calibration it's using? It should be listed on any panel of the Capture page, if you could post the entire page screenshot. Thanks!

Stoft1, 2016-02-12:
Because I was able to run chrome, the listed Cal has changed. However, in Firefox, it is nearly impossible to get a new cal to show (the Switch function fails most of the time). I did finally get the new CFL2700K cal file to 'stick' but the defect remains the same -- only a display up to 520nm:

[image]

jywarren, 2016-02-12:
Hmm, it's odd; I can't seem to reproduce this in Firefox 43 on windows; see screenshot calibrating with the same spectrum as you:
[image]

Can you try running it in Private mode in Firefox to ensure it's a clean copy of everything? And just to confirm, you are using a standard v3 Public Lab spectrometer webcam?

Thanks for your patience, we just have to narrow down what's causing this and it'll be a simple fix.

Stoft1, 2016-02-12:
Right, ran it in Private mode, fresh login to SWB; same issue. Here's a clip:
[image]
Doesn't matter what cal file is selected. I tried adding the search line (yellow) but it seems also unaware of any signal. Tried scaling but could not find a value that showed real data, just noise.
Yes, this is the PLab V3 'gumstick' camera from the proto I'm using built from the parts from Mathew. The camera seems readily detected and the image appears the same as it does with my matlab image utils. That suggests that the PLab end must be getting the source data which suggests it is all an issue with what is presented. The scale is shown as if there was a cal -- never a 'raw' scale -- so I'd guess the code is in that mode even though the exact cal file doesn't seem to matter at all. Since the display data stops at 520nm (scale point), I'd guess that would actually be 840nm data -- perhaps an offset of 320nm -- except that I'd have expected to at least see the spike in the red region. That suggests that the data is not from where the 'pixel line' was set within the graphic camera image. A mystery, but maybe some clues there.

jywarren, 2016-02-12:
I believe the calibration selection issue is resolved in #327 and this should help us narrow our attempts to reproduce this issue. Can you confirm that #327 is fixed and try once more please? Thank you for your patience and support.

Stoft1, 2016-02-12:
Ok, yes, I can now select the cal file (although doing so forces a complete recycle though selecing the camera and refreshing the image window -- though that may just be required as a byproduct of making the cal file switch). Yes, the CSV files now have the wavelength data. Thanks! for those.

Stoft1, 2016-02-12:
I just did an experiment by saving the 520nm-limited spectrum shown by Firefox 44.02: https://spectralworkbench.org/spectrums/71419# Thought it might give some additional clues. No wavelength data in the CSV file but the data appears to match what the plot shows.

jywarren, 2016-02-12:
OK; and you're running this with no calibration chosen during capture? Can you try to do that if you haven't since the fixes: https://spectralworkbench.org/capture?calibration_id=calibration

Stoft1, 2016-02-13:
Right, tried it again. It doesn't matter; no cal, or any of the cal files listed. Same problem with display stops at ~520nm. The 'black' area, which shows the spectrum color time series, does 'sweep down' as if it were showing the active spectrum, but it's all black (not even snow) so that function isn't sourcing any data -- though that display function is working.

Ok, New Observation. I'd been setting the line position manually, so I tried the "autoSet" button. SWB re-positioned the line about 40% of the way UP from where the bands appear in the image; black backround. (Now, I know that the position of the colors in the frame is about right as it appears and that the auto-set line position is wrong relative to the image frame.) However, the CFL specrum curve then appears! ... so the line position doesn't match the frame data:

[image]

Well, the plot is still squashed (yes, not shifted but scaled down into ~half the  display space) but at least I see something, so I set the cal reference to my CFL2700K and then save the result and hit the Calibrate button after the save and the plot appears. You can see the result here:

https://spectralworkbench.org/spectrums/71424#

Now, the cal appeared to complete but the display is still mucked up but I suspect that is from the same problem -- fix the original issue and all else will likely follow. Hope this jogs some memory of where to look in the code.

jywarren, 2016-02-17:
Hi, Dave - can you click "Scale video" as shown here, and enter `1` and tell me what happens?
 
[image]

Not sure if this is it, but I'm curious because i was able to use that button to reproduce what you're seeing -- a cut-off edge, ending at ~520nm -- by entering 0.5 in the video scale. I'm not sure why it'd persist over pageloads, but it's worth trying:

[image]

Stoft1, 2016-02-21:
Jeff, I finally tried the 'scale' factor. The sequence: acquire camera, shows spectral band of cfl, click auto-align for the pixel line, line jumps up to blank space but spectral curve appears, click to view spectrum, still left-shifted and 520nm cutoff, scale 1x, nothing happens, scale 2x and I get this:

[image]

So, I again scale to 1x and get this:

[image]

Note that the "2x" scaled waveform remains as the new curve is only plotted up to 520nm so clearly the display area > 520nm is not being overwritten.

jywarren, 2016-02-22:
OK, we can fix the not-overwriting issue. But when it's 2x, it is working
properly? If so, I then have to figure out how yours is stuck at half the
horizontal scale. Can you save during 2x capture and point me at the
spectrum so I can look at the image data?

Thanks!

Stoft1, 2016-02-22:
A scale factor of 1.5x seems closer. I saved a spectrum at: https://spectralworkbench.org/spectrums/71991# but even though I specified the CFL2700K cal file, that cal was not applied -- another bug. Also, as an aside, spectra naming is quite restrictive and, if you pick a bad character type, you get kicked out of the spectral page to an error page rather than just being notified to fix the name. The error message is also misleading as it says the spectrum has a problem -- when it is really just the naming of the spectrum. (Can't use a '-' or a '.' or spaces ... and there may be more...).