publiclab/spectral-workbench #317: Cannot access webcam from spectralworkbench

dcanprdn, 2016-02-05:
Username: dcanprdn
     I have been developing a vis spectrometer for absorbance measurements.  Spectralworkbench has been useful in development.  Evaluation of a given design change has been evaluated by CFL spectrum calibration.  Suddenly today (2/5/16) the webcam in my apparatus cannot accessed.  There was an icon below the spectrum picture on the capture page that would be clicked to change the camera from computer to webcam.  That is no longer present.  
    I have tried rebooting, cleaning the cache, history etc.  I have the camera icon in the chrome address bar and when I click on that and block the computer camera and then click and apply use of the webcam there still is no change and the computer camera image persists.
     It seems spectralworkbench cannot connect with my apparatus in anyway and the image from the computer camera persists.  
Using chrome, windows 10.  Cannot access webcam from this page\
https://spectralworkbench.org/capture

jywarren, 2016-02-05:
Hi, dcanprdn - sorry, we can add that back in, but Chrome started serving
us a notice that that method of changing webcams has been deprecated and
will be removed from future versions of Chrome. I wasn't aware anyone was
using it, but thank you for reporting this.

When you use the URL bar to change cameras, you do have to reload the page.
Does this make that work for you, or shall we revert the removal of the
button?

jywarren, 2016-02-08:
Hello dcanprdn - what do you think, should we revert that change or are you able to change using the URL bar?

dcanprdn, 2016-02-09:
Jeff

      Sorry, after your rapid response I have not responded quickly.  I do
not feel I have worked out the optimum approach yet but will give an update

     I superficially tried your suggestion with chrome but was not
successful however, I will look again more thoroughly.

    I installed Opera but was not able to get it to recognize my webcam in
my diy spec.

I installed firefox and it seemed to work however, it does seem to “get
confused”.  When I load the capture screen sometimes it shows a camera icon
beside a firefox icon top, a little right of center; also a camera icon to
the left of the url address space!  Upon clicking on the camera icon it
gives a choice of sharing your/my camera or not sharing your/my camera.
There is no change to the spectra; I just see my own image of course

     However, I have been successful at least half of the time indicated by
being given a choice between a HD camera and a webcam.

    Then when clicking on that I have a list of four “interesting” double
negative and double positive statements relating to camera, webcam use.  So
on clicking on one of those I could finally see my spectra.  Now I cannot
remember which one (smile)!

   Somehow today so far I have not been successful in getting the
spectralworkbench to see my webcam output.  Just seeing the camera.  Also,
today after signing on to spectralworkbench (secondary to public lab) and
confirming my identity it takes me to the capture page but there I am not
signed in?  So I sign in again and it shows my sign in and then I can click
capture and get to the capture page.

   Go figure?!

     Sorry that my response is convoluted but that is where it stands right
now.

I recall previously that chrome was recommended and firefox was not
recommended.  Interesting that things are reversed.

    I will keep working on it but am busy trying to reduce the focal point
of my modest (~ $ 3.00) webcam (Beijing special).

    I and a colleague are focused on making a relatively cheap spec for
hands-on (and hopefully brain on) student experiments that can be used to
determine spectra and absorbance values for given chemicals and
biochemical.  The focus is on inner city students! So far the cost stands
at ~$12.00 which is where we want to be!

    We wanted to use the raspberry Pi with its camera (a much better
camera) but that setup especially with the camera and SDcard did not seem
robust enough (upon reflection) for such a learning environment.

    So I am hoping use of our system with a cheaper webcam coupled with
your software will allow a cheaper alternative.  Maybe ultimately a
student/teacher page would facilitate classroom  use if that is ok with you
folks.

    I appreciate that more of the workings of your site in terms of data
files is being made more accessible because we have to access the data
output from the CMOS sensor at some level to get the wavelength vs R, G,
and B data and put it into an excel file and both plot it (W) and use that
data (at the maximum absorption value) to determine absorbance and hence
concentration for a given compound.

    Rest assured I will keep working on this situation and will report back
to you when I can hopefully be clearer.
Dave P

jywarren, 2016-02-09:
Hmm, thank you for your thorough response -- if you're able to take screenshots of the state when you first log in in Chrome, including the URL bar, perhaps after clicking on any camera icon you see there first, that'd be really helpful. Chrome is definitely working for many/most people, so I'm interested to ensure you're able to use it too.

Thanks!

dcanprdn, 2016-02-26:
Jeff Warren
    Jeff I have been sidetracked but appreciate your support previously.  I
was having trouble using spectral workbench calibration with chrome.  I
never could get chrome to work even when prompted with a reload as you
suggested.  However, opera and firefox work great and I can move on now.
Thanks
Dave P

jywarren, 2016-02-29:
OK, glad to hear! Thanks.

closed 2016-02-29 by jywarren