publiclab/spectral-workbench #323: "save with selected calibration" requires you to refresh the page instead of just showing the spectrum with the selected calibration

ethanbass, 2016-02-11:
Not such a big deal once you figure it out, but you have to refresh the page after saving a new spectrum in order for it to appear with the calibration... I tested this bug in firefox and in chrome on OS X maverick.

jywarren, 2016-02-11:
Hmm, really? Are you sure it doesn't just take a little while? Is there any
way you can screenrecord this so I can try to reproduce it? Thanks!

ethanbass, 2016-02-11:
do you mean a video? or a screenshot?

jywarren, 2016-02-11:
Well if a video is a pain, a sequence of screenshots I guess? I'm just
trying to get as much information as possible.

ethanbass, 2016-02-11:
[image]

Here is a screenshot of what i'm talking about, this is after about 5 minutes and the browser doesn't appear to be loading anything. I don't know what time zone the app is working in. It says 23:01, but I think I uploaded the spectra at 5 pm my time and the picture as you can see if from 5:05. You'll notice that it says under notifications "this spectrum needs to be calibrated" but you can also see that the calibration operation has been completed.

When I refresh the page and take another screenshot, the calibration snaps into place and the notification goes away, as you can see below.

[image]

jywarren, 2016-02-11:
Ok thanks!!! Can you open a new issue for the timestamp though?

Thanks.

jywarren, 2016-02-12:
Can you open a new issue for the calibrated comparisons too? Thanks, Ethan!

ethanbass, 2016-02-12:
yeah

jywarren, 2016-02-12:
Awesome, very grateful. As to the above screenshots, it looks like the calibration is being generated, but it's not refreshing the graph once it's done. Thanks for giving it some time to see if it goes through. 

Reproduced; the log shows: https://gist.github.com/jywarren/3995e6416a6a69fdb34f

following up...

jywarren, 2016-02-18:
Checking if this has been resolved in v2.1.9...

jywarren, 2016-02-18:
I believe it is -- @ethanbass, can you try it out? Thanks!

closed 2016-02-18 by jywarren