publiclab/spectral-workbench #324: time stamp bug

ethanbass, 2016-02-12:
I took this spectra on Feb 11 at 5 pm but the time stamp says 23:01. The time stamps on all of my spectra taken in spectral workbench 2.0 seem to be offset by 6 hours. 

[image]

jywarren, 2016-02-12:
Thanks; this page has some resources on determing user timezone:
http://stackoverflow.com/questions/4269270/rails-3-setting-the-timezone-to-the-current-users-timezone

jywarren, 2016-02-19:
We can use http://momentjs.com/timezone/ -- we're already using moment.js, actually!

jywarren, 2016-02-19:
Hmm. I'm a bit loathe to introduce a whole new library for this, or to execute further javascript just to parse a date into a local timezone. It kind of sucks, but I'm just going to add "UTC" to the end of that timestamp. If you feel strongly about it, please speak up and we can do something more.

commit 947a80c1 (2016-02-24): added UTC to timestamp for spectrum, close #324
closed 2016-02-19 by jywarren