publiclab/spectral-workbench #325: comparison feature calibration issue

ethanbass, 2016-02-12:
Also, a bigger problem i've just found, which is probably related: I have two calibrated spectra from the same session (i.e. I went back to the same window of spectral work bench after saving one spectrum and saved the next spectrum). The spectra are both calibrated, but when I go to compare them, the one I am trying to overlay is not calibrated. It doesn't matter what the order of operations is (I can pull up either spectrum and try to compare it to the other and the same problem arises).

I am attaching a screenshot of two spectra superimposed, where one is not showing up as calibrated. Perhaps it has something to do with the new operations function. I can only guess that the comparison function may not be accessing the "snapshot" where the calibration information has been saved.

[image]

jywarren, 2016-02-12:
Thanks, Ethan - can you give me spectrum IDs to compare? Thanks!

jywarren, 2016-02-12:
Reproduced and confirmed for https://spectralworkbench.org/spectrums/71375 and https://spectralworkbench.org/spectrums/71376 -- will look at this tomorrow morning. Thanks again!

jywarren, 2016-02-18:
`calibrate` operations weren't properly storing a snapshot; this is resolved but you have to delete and regenerate the operations. I've tested this on forks: https://spectralworkbench.org/spectrums/71811 and https://spectralworkbench.org/spectrums/71810

However, now that I've forked your spectra, you won't be able to modify them. You can either fork your own, or if you like I can delete my forks.

Anyhow, this bug is resolved in v2.1.9, now live! Thanks, Ethan!

closed 2016-02-18 by jywarren