publiclab/spectral-workbench #335: set precision throughout spectrum/snapshot saving

jywarren, 2016-02-18:
@Stoft1 - can you offer insight on what precision we should store in any spectral data we keep? I'd like to just standardize this and get it right. Thanks!

Stoft1, 2016-02-18:
It is common for lab measurements to provide greater reported precision than the actual level of accuracy. I've not been able to extract better than 2nm FWHM measurement values from my prototypes even when using 1024 pix width camera settings. This suggests that the default 640x480 webcam data, at best, is probably closer to 3nm FWHM resolution. Since most spectra are collected at only a little better than 1nm per pix (a calibrated 400-700nm 300nm span is only about half the 640 pix width), reporting anything more precise than 1nm is misleading (and reporting 534.075nm in a data file is VERY misleading).

1) If you kept the original pixel data, then there is no opportunity for data loss of resolution
2) Fractional wavelength values have no meaning in a >2nm precision world
3) 1nm integer data / plot resolution is legitimate as long as the measurement error is noted
4) 1nm integer data is so much easier to display, plot, handle, etc.
5) Therefore, all you need are 2 units: pixels and nm -- both integer values

Admittedly there is some one-time error in scaling from pixels to nm but since there are slightly more pixels than integer nm, you will never be off by more than 0.5nm (and usually much less) within a measurement accuracy of 2-3nm. Also remember, that many spectrometers calibrate with CFL data which does not even resolve the double green Hg line which likely adds several nm of additional error. There is additional error which results from where the pixel line is drawn through the spectral band (as per my other recent notes). I'm still working on that issue in search of better automation to capture a consistent spectra.

Magnitude is also a consideration. The camera has 8-bit dynamic range while the plots can show relative percentage of max. I'm convinced that there is sufficient noise and measurement repeatability issues such that one cannot claim 1/256 intensity accuracy; just 8-bit resolution. Converting to % could provide data storage of (eg. 89.7%) 1/10% resolution (4x the collected data resolution) and be assured of not losing information but, again, the measurement error is much worse than the numbers suggest. Just keeping the 8-bit values and always just scale from that data to get % displays would be fine. Keeping integer % values would not retain precise measurement values, but would not overtly affect the real measurement error. Still, I'd opt for just retaining the 8-bit data and always just scale for display purposes.

In short, you could keep 1) original pixel and 8-bit data and 2) calibrated 1nm resolution CFL-cal'd wavelength data with 8-bit intensity. All else can be easily extracted from that.

It could be argued that once a spectra has been calibrated the pixel data need not be kept. The counter argument is that the cal bight have been done wrong so the original pixel info must be kept. The reply argument is 1) commercial instruments don't generally provide raw data (at least, not easily accessible) and that 2) allowing re-cal of data opens the process for additional errors. If keeping raw pixel data is easy and cheap, then it's probably a small issue.

Hope this is helpful.

jywarren, 2016-02-19:
This is very helpful. I want to start with intensity first, then move on to wavelength -- mainly because we're seeing data saved like:

    {\"average\":28.05,\"r\":30.599999999999998,\"g\":17.85,\"b\":38.25,\"wavelength\":569.957},

I also want to separate the discussion of what is within the detection precision of the hardware from what precision the software outputs given an input of a given precision -- we can deal with hardware limitations later, and we should -- but the software should at a minimum not insert significant digits where they weren't reported. 

By using intensity as our first case, we have a clean example of what's reported and what its precision is. The output should have no greater precision than that. 

Anyhow, now that I look them up (first time since 8th grade!) the rules of significant figures and [significance arithmetic](https://en.wikipedia.org/wiki/Significance_arithmetic) are fairly complex, I think we need some guidance on how they might be implemented in JavaScript, and if and how they might overlap with floating point math issues -- not that we're going to be that precise, but just to be thorough. 

I'm seeing a few resources we might use:

* https://github.com/k4kfh/sigFig.js - a library to handle significance arithmetic -- although, given its syntax, it seems like to require use of this in any data-handling code would be pretty onerous...
* http://mikemcl.github.io/big.js/ - an arbitrary-precision decimal system for JavaScript, which would avoid floating point issues (and likely clean up the 30.599999999999998 above)
* https://github.com/MikeMcl/big.js/wiki says decimal.js (same author as big.js) is more suited to scientific applications and explains why: http://mikemcl.github.io/decimal.js/

One issue is that according to the guidelines of significance arithmetic, we only want to round to significant figures once, just before displaying the data. I'm not sure how that maps to a more automated system such as we are using.

But maybe we can just talk about this simply: our intensity values are 0-255, so any value we display must have (if I'm doing this correctly) a maximum of 3 significant figures. 

So, we could just round to 3 sig figs immediately before displaying a graph, and immediately before saving data. Which would be a good start -- what do you think? @Fastie - any thoughts on this?

Fastie, 2016-02-19:
I am not quite up to speed on this issue, but it sounds like rounding to three significant figures of intensity before displaying or outputing data makes sense.

With wavelength data, the limitations of the hardware (the camera) and the software (SW downsamples every input image to 640? pixels) makes Dave's idea of focusing on integer nm or pixels reasonable for now. But simple spectrometers have the potential to capture much higher wavelength resolution than that. If you have a solid setup and use a good camera, photos of diffraction patterns can probably have 10 or 20 pixel per nm precision. If SW wants to be capable of handling such better input data, it might want to preserve the option of dealing with a few more significant figures. Maybe in the future SW will no longer downsample all input images, so it will need to handle more than a few significant figures.

jywarren, 2016-02-19:
Yes, in fact I've laid the groundwork for higher resolution capture, and it shouldn't be difficult to enable it soon. Any solution we come up with here should be scalable.

jywarren, 2016-02-19:
OK, I'm creating a method to help manage significant figures:

````js
/* ======================================
 * Iterate through channel data and set precision of
 * each entry to sigFigures>, but defaulting to
 * _spectrum.sigFigIntensity which is 3.
 */
 _spectrum.setSigFigures = function(sigFigures) {
````

I'm going to run this just before refreshing the graph, and see how it goes in my development branch of code.

jywarren, 2016-02-19:
OK, I got it working both just before displaying the graph, and also just before uploading snapshot data.

So, I'm seeing this now:

    {"average":45.9,"r":68.9,"g":45.9,"b":76.5,"wavelength":392.6},

This is correct # of significant digits based on what I set (3 for intensity, 4 for wavelength, and we should discuss that too), but does it follow addition or subtraction rules too? Can you check my work on that?

jywarren, 2016-02-19:
Also, I'm using JavaScript's `toPrecision()` -- https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Number/toPrecision so maybe I don't need the sig figs JS library I listed above.

Stoft1, 2016-02-20:
I'd agree that initially a 3 significant digit value for intensity works with an 8-bit depth. However, I also agree with keeping flexible for the future (i.e. say 10-bits). That said, I think it should be easy to detect that the input data is 8-bit vs 10-bit. Recording 130 (50.78 %FS) could be reported as 50.8 w/o loss of precision. However, 10-bits would require 4+ digits for full precision but you would not want to report 50.781 with an 8-bit system as the added resolution is just numerical noise (as in comparing or combining data from separate hardware).

There is yet another issue with the wavelength data. If the greater-than-1nm precision is generated, logged and reported from individual devices and each has it's own 0.X increments of the pixel:nm relationship, then comparisons, and even plotting, are made more difficult -- an added expense of complexity which, again, with 8-bits and a 3nm FWHM measurement resolution, buys you nothing. Say one system calculates a pixel data point at 456.3nm and another reports that same wavelength pixel at 456.8nm (in reality, I suspect the hardware will actually be worse than this) then you have added complexity to decide which points correspond between data sets. Yes, I know this is not of important practical concern with PLab devices, but I mention it just to point to the advantage of getting the display and reported data (nm and %) using common intervals (i.e. integer nm and 0.X%).

Again, since I believe it is easy to detect the camera resolution, the "raw" data is pixel data and is easy to save. The CFL(or other) wavelength correction is easily 1) saved as "raw" pixel data as well and 2) easily used to interpret a spectra's "raw" data. Both the display and download data is simply auto-generated on-the-fly from the "raw" data whose accuracy is integer pixel data. This also frees the display scaling from the pixel level.

Remember, even a narrow 3-pixel peak can have a 1-pixel jitter with very stable hardware. So, save pixel data and display or report with the number of significant digits which just cover the source resolution. For the graphic display, it is simpler to plot with integer nm and integer % values for the X / Y axes and then let the plotting do the interpolation work using the same resolution that represents the "raw" pixel resolution.

jywarren, 2016-02-22:
Actually, the system is currently set up to deal with comparisons, etc, of data with differing resolutions. That's another issue I'd like to discuss -- maybe on the spectrometry mailing list? I'll start a thread, but the gist is that we don't have to worry about that in this context at the moment.

Agreed on your second point re: raw data - we always keep the image data, and due to the snapshots system, we keep various older versions of data before additional processing. So no worries there.

What about our wavelength significant figures? I chose 4, but upon what do we base our decision for wavelength sig figs? Using the two multiplication/addition rules, and the actual math we do to calibrate? And the precision of our reference? We're using 5 sig figs for our blue/green calibration references. And roughly, we're (using [this example](https://spectralworkbench.org/spectrums/71812)) getting 1.0791 nm per pixel with the v3 spectrometer.

Stoft1, 2016-02-22:
The old phrase is "Measure it with a micrometer, mark it with chalk, then cut it with an axe."

You have three data stages of interest: 1) raw pixel intensities with noise, 'position' jitter and other errors, 2) what you report in a data file with nm units and 3) what you display. As long as the calculations between these stages do not diminish the accuracy or resolution of the source data, you are fine. The issues is what resolution to use in resultant data where units have changed.

I'm of the opinion that the result data (data file) should not lose resolution but also should not increase (unrealisticly) the resolution relative to the source. Display data should reasonably show the source resolution but no more than that (no interpolation as that 'data' is not real).

EG: 8-bit, 640pix image: 256-level data raw data, convert to % means NNN.N digits to not lose resolution but NNN.NN would "add" bit depth resolution that is not there. Splitting hairs that 129/256 is 0.5039 so 50.39% means the extra 0.0X digit is not justified because the source error swamps that added resolution from the units conversion. So, just keep the 50.4 . (This means a range of 100 (256/256) -to- 000.4 (1/256) ). Having fewer pixels than nm forces an approximation which means that periodically (about every 7th) there will be two pixels which "overlap" for being the "correct" value for a specific wavelength nm integer value. When the nm:pix ratio is so close, the pixel data error will swamp the conversion math so for display, just keep integer nm values from the conversion. For the calculation resolution, you could use 0.1nm resolution. However, reporting 0.1nm resolution in a data file (for download or comparison) seems both unwarranted and hold more complexity that does not provide added benefits; the raw data errors swamp the 0.1nm resolution. So, I do not see rationale for using more than 1nm integer wavelength values.

For greater than 8-bit and 640pix depth and resolution, this same analysis can be used.

jywarren, 2016-02-24:
OK, just getting organized; this issue will be the discussion for stored precision,

#340 will address data interpolation, which is between Dave's #1 and #2 (in-process handling of data), and

#344 will address **display precision**, which I agree is important but is implemented in a separate part of the codebase. This also includes comment by @Stoft1 relevant to storage precision (a comparison of storage vs. display precision).

jywarren, 2016-02-24:
And, excellent -- I'm comfortable assuming 8-bit until we get a request for higher intensity precision. I just understood that 0-255 means 000.4-100.0, or 4 sig figs. I was assuming it'd be 3, but that makes sense. 

And I think I understand that your comment (which I moved to #344) means 4 sig figs for stored wavelength resolution (skipping for now the "E" digit, as you referred to it). 

So, super, thanks! 

I'm going to merge in 4 sig fig code for both intensity and wavelength stored precision, then we can move on to display precision in #344. Please re-open if I'm mistaken!

closed 2016-02-24 by jywarren