publiclab/spectral-workbench #339: reduce noise with time averaging

robotp, 2016-02-19:
-Add more than a single webcam row
-Add several frames together, such as co-adding rows of the curtain plot shown above the plot
-improve dynamic range during/after averaging to larger than 8bit

jywarren, 2016-02-19:
We'd like to do this, and have in the past, but we'd like to be able to
warn people that this won't work for curved spectral images, only
live-captured ones. We can tell to some degree from tagging, and we can
simply show a visual warning, but it'd have to be done right.

Thanks for the suggestion!

jywarren, 2016-02-19:
Also, are you familiar with JavaScript coding? The new API is great, and we're seeking new contributors.

robotp, 2016-02-20:
I have used some js in the past; I can give it a shot!  I am not totally sure how to implement it in the browser.

robotp, 2016-02-20:
Even if we did not use multiple webcam rows, control over time averaging would allow the user to get SNR to the level they desire.
Thanks for the help.
Also do you have a suggestion on a webcam that will go further into the NIR.  I have removed the optical filter from the "gumstick" cam that come with the pubLab spec 3.0, but still get cut off at about 775nm

jywarren, 2016-02-20:
I'll be posting some more docs on how to use the API soon, but here's a
starting point:

https://publiclab.org/wiki/spectral-workbench-api

This method gets you a single line of image data:

https://github.com/publiclab/spectral-workbench/blob/master/app/assets/javascripts/spectralworkbench/SpectralWorkbench.Image.js#L52

Regarding the webcam, the best place to ask is the spectrometry discussion
list!

jywarren, 2021-07-27:
I think a button in the capture interface that averages the last X frames is maybe the best way to do this?