publiclab/spectral-workbench #353: Standardized WebRTC camera selection

jywarren, 2016-04-12:
Reopening since #317 -- Chrome has changed their WebRTC camera selection API, but they have a working demo here:

https://webrtc.github.io/samples/src/content/devices/input-output/

This is confirmed to work in the latest Android Chrome and Android Firefox, for front and rear facing cameras. It says:

> Get available audio, video sources and audio output devices* from mediaDevices.enumerateDevices() then set the source for getUserMedia() using a deviceId constraint.

Demo code here: https://github.com/webrtc/samples/tree/gh-pages/src/content/devices/input-output/js

Sarah-C, 2018-05-15:
Please!  I'd like to select my rear camera now1

jywarren, 2018-06-04:
If you're in Android you can do this in Firefox mobile. But yes, i agree, this would be really really helpful!!!

jywarren, 2018-06-04:
I added `help-wanted` tag to see if we can get someone to try and solve it!

jywarren, 2019-03-28:
See https://github.com/publiclab/spectral-workbench.js/issues/87 for more on this!

jywarren, 2021-09-23:
Done!!!