publiclab/spectral-workbench #361: The overlay measurement is gone

YogiBGitHub, 2016-04-25:
When pointing to the graph on a saved spectrum, the measurement is no longer visible. Tried two different browsers, Chrome and Edge. Not tried a different computer though.

Username: YogiB

jywarren, 2016-04-25:
Hi, do you mean the little tooltip/popup that follows your mouse? Confirmed
this at https://spectralworkbench.org/spectrums/55073, thank you for
reporting.

jywarren, 2016-04-25:
Confirmed that this is not working on SpectralWorkbench.js either: https://publiclab.github.io/spectral-workbench.js/examples/

Probably related to stylesheet classname changes. Also the image itself is not properly masked.

YogiBGitHub, 2016-04-25:
Yes, that’s correct. It is very difficult to measure the exact peak wavelengths without this little helper.

jywarren, 2016-04-25:
Sorry for the bug! I have it on my schedule to look at tomorrow morning.
Thank you for your patience.

jywarren, 2016-04-26:
I think this is fixed, I'll publish the change in the next 10 minutes. Can you confirm then?

closed 2016-04-26 by jywarren

jywarren, 2016-04-26:
Sorry, please hold, still figuring this out.

jywarren, 2016-04-26:
OK, this is published now. I've confirmed that it works in a recent Chrome; does it work for you?

YogiBGitHub, 2016-04-26:
Thanks Jeffrey. Now everything seems fine, also in Chrome.

 

No virus found in this message.
Checked by AVG - www.avg.com
Version: 2016.0.7539 / Virus Database: 4563/12108 - Release Date: 04/26/16

jywarren, 2016-04-26:
Super! Thanks.

commit 7120f345 (2016-04-26): new version of spectral-workbench package, fixes #361