publiclab/spectral-workbench #365: uploading on CSV (beta)

dhaffnersr, 2016-04-27:
## What happened just before the problem occurred

hey Jeff, i see what my problem was on uploading the csv raw data, i needed to place a comma after each wavelength number. Well I am politely inclined to say, that is wayyyyy too labor intensive if I have almost a hundred numbers to place comma's after, with the csv files you requested I have 3 files, that's an awful lot of commas to place.

There has to be a better way.

## Relevant URLs

https://spectralworkbench.org/upload#

## PublicLab.org username

dhaffnersr

## Browser, version, and operating system
google chrome windows 7 (ultimate)

For bug reports, fill out the above template; for feature requests, you can delete the template.

jywarren, 2016-04-27:
Hi Dave! It sounds like you don't quite have a standard CSV -- that stands
for comma-separated values. But I believe if you open it in excel you can
save it again with commas instead of spaces.

If that works, do you think you could add that as a tip to the part if the
wiki that talks about uploads?

dhaffnersr, 2016-04-27:
Ok Jeff, let me try that and I'll let you know how it goes, ya learn
something new everyday!

jywarren, 2016-04-27:
I added your note to the instructions linked to from the upload form. Thanks!

closed 2016-04-27 by jywarren