publiclab/spectral-workbench #369: Every spectrum I take is 100nm off...

dhaffnersr, 2016-05-10:
## What happened just before the problem occurred
 I was doing a CFL calibration like I did this morning and all of a sudden when I go to calibrate the wavelength is ahead by 100nm? I tried a 405nm uv laser and green laser and it does the same?

## Relevant URLs

https://

## PublicLab.org username
dhaffnersr

(to help reproduce the issue)

## Browser, version, and operating system

For bug reports, fill out the above template; for feature requests, you can delete the template.

jywarren, 2016-05-11:
Hmm, can you link us to one that is incorrect please?

jywarren, 2016-05-11:
And did the issue begin yesterday morning? Can you see in your recent
spectra when the error began to appear, precisely? We haven't published any
changes in the past few days, so I'm trying to narrow it down. Which was
the first spectrum that showed the error?

Thanks!