publiclab/spectral-workbench #385: Compare does not work

YogiBGitHub, 2016-11-14:
I don't think compare does not work any more. When selecting the 2nd spectrum, the 1st is just replaced with the second one. In fact, to be honest if I dare, I miss the earlier interface (legacy 1.0), where one could compare many spectrums, and even equalize the height for a better comparing. It still works, but the calibration is gone. Any chance of getting the calibration to work with the legacy 1.0?

jywarren, 2016-11-17:
Hi, apologies, we just wrapped up the [Barnraising](https://publiclab.org/barnraising), and coming back online now. Can you link to the spectra you're working with? Thanks.

YogiBGitHub, 2016-11-17:
Hi, sorry, but I might have been clicking on the link instead of the apply button, when comparing. Because now it works. But I can only compare two spectrums at the time. When trying to add a third, nothing happens. And I also miss the larger vertical area from the previous version. The vertical resolution would improve if the spectrum could be stretched vertically.

jywarren, 2016-11-17:
Hmm, odd -- well ok but let's check that there's not a problem with more
than one comparison; can you link to your spectra so I can give it a try?
Perhaps we could write a test for it.

YogiBGitHub, 2016-11-20:
Ok, here are three spectra:
https://spectralworkbench.org/spectrums/91798
https://spectralworkbench.org/spectrums/91778
https://spectralworkbench.org/spectrums/90483
It does not matter which spectrum I use, when clicking apply on the 3rd spectrum, it is not added. I am using Chrome, but have not tried another browser yet, or another computer.

jywarren, 2016-11-22:
Hi, indeed I also only see one comparison; the most recent. But if you switch over to the Comparisons tab, you can see that multiple are listed. 

So this is a bug, but if you press `Save as a set` below the list, you can see them all in the same graph, so that's a workaround for now. Thanks!

YogiBGitHub, 2016-11-22:
Thanks. And thanks for the workaround tip. From time to time I do some programming myself at work (.Net & C#), and know the devil called bug quite well. Hmm, just doing a small change in this module, should not affect anything else…. Well, it’s tiresome to apply zillions of “if tests” for conditions that never should occur. But they will. 

 

And user input can be hell. A customer claimed about error in a web-form. It worked well for everyone else. I discovered he typed “approx. mid May” in the date field. So validation was added.

 

No virus found in this message.
Checked by AVG - www.avg.com
Version: 2016.0.7924 / Virus Database: 4664/13457 - Release Date: 11/22/16

jywarren, 2016-11-22:
> gaJpZM4Kxlqd> . <https://github.com/notifications/beacon/ANHaefM7d5Yl2D_

😀