publiclab/spectral-workbench #391: The wavelength is null in csv, json, xml

bosonbeard, 2017-08-16:
## What happened just before the problem occurred
Smartphone spectrometer, manual uploading, manual calibration
After calibration in the files for downloading (csv, json, xml) , the wavelength is zero(Null). On the chart, everything is correct. Similar already had a couple of years ago in the old interface.

## Relevant URLs

https://spectralworkbench.org/spectrums/111365

## PublicLab.org username
bosonbeard
(to help reproduce the issue)

## Browser, version, and operating system
MS Edge, SrWare Iron (chromium) 58.0.3050.0 , OS Win 10

For bug reports, fill out the above template; for feature requests, you can delete the template.

bosonbeard, 2017-08-16:
I realized that this depends on the length of the image, cropped the image in another version and the error was gone
https://spectralworkbench.org/spectrums/111367

jywarren, 2017-08-21:
Ah, was it that the original image was too large? I think the JSON encoding of the data could exceed the available space in the data storage column we use... but if so we should return a related error.

bosonbeard, 2017-08-26:
I do not know what is the cause of the error, but this problem occurs quite often. Here is another example
https://spectralworkbench.org/spectrums/98536

jywarren, 2017-12-07:
I believe this may be related to #394 with a pending fix in #395! Merging to track together.

PR #395 (2017-12-07): Expand snapshots with higher data limit to remove errors
This should fix #392, fix #391, fix #394, and maybe more. Starting with a test.
closed 2017-12-07 by jywarren