publiclab/spectral-workbench #399: Training a Model Based off Spectral Workbench Dataset

Lucaszw, 2018-04-27:
I think it would be really cool to have predicted labels for spectral data based off a pre-trained model from Spectral Workbench's dataset.

I imagine something like this would be awesome: 
https://transcranial.github.io/keras-js/#/inception-v3

Where you have both the confidence scores while also a heatmap to show which parts of the spectrum were most relevant to the particular class. 

Something like the image blow:

[image]

I am thinking it would be nice to develop this using TensorFlow.js

Lucaszw, 2018-05-10:
For such a project there will likely be two major components:

1) Creating a labeled dataset (which will require making some decisions as to which tags are the most common, or perhaps which spectral patterns are the most diverse).

2) Training and distributing the model for inference. With Tensorflow.js , we won't need a server side api for inference (since you can do this on the client side). Tags of similar spectral patterns, and heat maps I can see being very popular.

jywarren, 2018-05-15:
Just to clarify, would the workflow be to feed Tensorflow a big set of
labelled spectra, then it generates a model or result dataset which can
then be used to classify new data? I'm just trying to think through this
but where would we want to try to store the resulting model?

Thank you!!!

Lucaszw, 2018-05-18:
Yes exactly! The dataset and model(s) could be hosted on any server really.  The model can then be loaded directly via a URL (and cached in the users web browser ).  See: https://js.tensorflow.org/tutorials/import-keras.html

jywarren, 2018-06-11:
I was wondering if it'd be possible to build up a model by pointing the
spectrometer at stuff -- like, scan it in by scanning over an example
material to collect data... what do you think?