publiclab/spectral-workbench #410: page load time

hedbergj, 2018-10-11:
## What happened just before the problem occurred

The page can take a long time to load, and students are complaining about the page and analysis processes generally being very slow, which I can confirm. (min 5 seconds for just the page load, sometimes much longer) Should I investigate hosting a local version somewhere if there are no plans to improve the server performance?

## Relevant URLs

https://spectralworkbench.org/

## PublicLab.org username

(to help reproduce the issue)

## Browser, version, and operating system

chrome, 69.0.3497.100, osx 10.12.6

For bug reports, fill out the above template; for feature requests, you can delete the template.

jywarren, 2018-11-01:
Hi! Sorry, i was traveling at the time this was posted, but we are looking into this now. I think when lots of people simultaneously upload, things can get a bit slow. Apologies! We are looking into ways to optimize now.

jywarren, 2018-11-01:
Could be worth installing Skylight on as well @icarito 👍

icarito, 2018-11-01:
Thanks for your report! I'll be reviewing the performance situation to discern what we can do. We've been wanting to establish a remote process queue to avoid overloading production web server with image processing.